open-ephys/plugin-GUI #533: LFP viewer 'Sort by depth' doesn't work for NP2 4-shank

LiesDec, 2022-11-17:
Hi,
When using the NP2 4-shanks the channels are not correctly sorted after applying the 'sort by depth' function in the LFP viewer. 
I would like to double check that this is indeed ONLY a display error and not something more fundamental and that my data is not recorded as I think. From my offline analyses I don't think so but I think it is good to be 100% sure :) 

For simplicity I'm using on both probes the same channel map and a channel map that only selects electrodes on one shank. In the image below you can see the Imro file plotted with the associated channel numbers. 
[image]

However, when I apply the _sort by depht_ function in the LFP viewer this is the sorting I get. 
[image]

As you can see this is not what you would expect from the imro file. 
Here is also the imro file I used: 
[fourth_density_shank_0.txt](https://github.com/open-ephys/plugin-GUI/files/10030455/fourth_density_shank_0.txt)

Together with @marinechap we've tried to look for the error in the high level display function but that seems to use correctly the y-pos. Can someone help us with finding the bug in the deeper layers of the code and double check it doesn't influence anything else?

jsiegle, 2022-11-18:
Thanks for reporting this! The bug is on [this line of NeuropixThread.cpp](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/ef49f8a1917ba336c67897ed37237a4d2a743604/Source/NeuropixThread.cpp#L1099). The step in which the depth metadata was extracted assumed that the channels are ordered within a bank, which is not the case for Neuropixels 2.0 probes.

I fixed it by [reading the selected electrode directly from the probe settings](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/4223183216a7cdab2c74c11f46a73b8a778ee6cb/Source/NeuropixThread.cpp#L1087). The `add-geometries` branch includes this fix, and also updates the signal chain after loading an IMRO file, so the new depths will propagate to the LFP Viewer. 

The good news is that the actual channel configuration (and channel metadata in the settings files) was correct; the problem was just with the depth info used by the LFP Viewer.

If you can double check it to be sure the behavior is correct, it would be much appreciated! After that we will merge this into `main` and put out a new release ASAP.

MarineChap, 2022-11-18:
Thanks @jsiegle ! I installed your fix on our computer this morning. We will give you a feedback soon.

jsiegle, 2022-11-18:
When sorting by depth, the shanks are displayed consecutively (all channels on shank 1 first, then all channels on shank 2, etc.)

Currently, you can use a vertical split in the LFP Viewer to view up to 3 shanks side-by-side. We are brainstorming better ways to visualize multi-shank data—let us know if you have any suggestions!