open-ephys/plugin-GUI #534: files not fully closed after stopping the recording

neurochen, 2022-11-20:
We are trying to automatically delete open ephys data after aborting a recording session. Previsouly in version 0.5.x, we could simply delte the folder after either manually or remotely stopping the recording. However, in 0.6.x, we can't delete the folder anymore as some of the files are still in use. The only way to delete it is after starting another recording. Could you please take a look at this? 

Please see below for the message.
[image]

medengineer, 2022-11-22:
Looks like the sync_messages.txt file wasn't resetting after recording stops. 
I've added a fix here:  [open-ephys-test.zip](https://github.com/open-ephys/plugin-GUI/files/10061413/open-ephys-test.zip)
Can you try replacing your open-ephys.exe with the linked one and confirm this solves the issue?

neurochen, 2022-11-22:
Thnaks. It worked now.

closed 2022-11-22 by neurochen