open-ephys/plugin-GUI #535: Hard drive recording saturation

guilhemibos, 2022-11-23:
Hi

I was recording today from 163 channels from 3 different areas (one with a 32 channel linear probe, 2 areas from 64 channels Utah arrays plus 3 analog channels). 

I've tried to record each area in a specific node so that Kilosort spike sorting will be easier. However, doing that saturated my hard drive writing capacities and recording stopped after ~1 hour. That was surprising because the day before, I was able to record from all channels from a single node for more than an hour without any failure. 

At first, I did it a dumb way. I've added splitters and after each splitters, I've added channel map in which I selected only the appropriate channels followed by a recording node for each (2 splitters, 3 channel maps, 3 recording nodes). Then, I realized it was not necessary, and I've added 3 successive recording nodes in which I selected the channels I wanted to record from. However, the issue was the same with both configuration.

It seems to work fine with just 2 record nodes though.

I'm wondering what is the best strategy:
1. record everything on the same record node (no problem so far) and separate offline the data by areas (splitting the continuous.dat file in 3)? In which case, what would be the best way? Would you be able to send me a function which would do that properly?
2. record from different nodes (two actually) which separate the one recorded from the laminar probe from the Utah arrays? In which case, do I have to do something special in my acquisition chain?

Thanks
Guilhem

jsiegle, 2022-11-29:
Hi Guilhem – each Record Node uses a separate thread, which increases efficiency when writing data across multiple drives, but likely leads to conflicts when multiple threads are trying to access different locations on the same drive.

Which spike sorter are you using? The one I'm most familiar with (Kilosort) allow you to load a channel map file that specifies the channel order and any channels to ignore. So you could sort the same .dat file containing data from 3 probes and analog channels with the irrelevant channels masked out. If your sorter has that option, that is the easiest way to solve the problem.

Alternatively, if you add two more drives (ideally SSDs) to your machine, using 3 Record Nodes should work well.

guilhemibos, 2022-11-29:
Hi Josh

Thanks for the clarification. We've opted for a single node. 
The final data file is quite big (45 to 60 Gb), so loading everything in a single sorter can be tricky. Moreover, while I use KS for 32 channels (linear probe), I record from 128 channels from Utah arrays and I'm not sure Kilosort is the best tool for that. So I'd like to try different tools, including offline sorter from Plexon (I've open a question for that here by the way: https://github.com/open-ephys/analysis-tools/issues/102), and loading such huge files in Plexon OS will be painful. We try to split the data in 3 (one for each area).

jsiegle, 2022-11-29:
Does Offline Sorter try to load all the data into memory at once? That is unfortunate.

In any case, splitting up a `.dat` file can be done easily in Python:

```python
import numpy as np
total_channels = 163
data = np.memmap(filename_in, dtype='int16', mode='r')
data = np.reshape(data, (data.size // total_channels, total_channels))
data[:, 32:96].tofile(filename_out)
```

This will write channels 32-96 to a new `.dat` file.

closed 2025-06-08 by jsiegle