open-ephys/plugin-GUI #539: Kwik files in file reader

tduvan, 2022-12-11:
Hi,

I'm trying to visualize data from experiments that are saved in Kwik format. I've tried gum versions 4, 5 and 6 but was unable to open the files. Then I've found out that there is a related plugin but when I open the install plugin tab nothing comes up in the list.

Could you help me with the issue?

jsiegle, 2022-12-12:
How were these Kwik files generated? The Kwik format has been deprecated for a while now, but if you saved a Kwik file with the Open Ephys GUI, you can *probably* read it back in with the same software version. If they were created a different way, then it likely won't work.

If at all possible, I would recommend extracting the data in Python or Matlab and re-saving it in [Binary](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html) format. The [`open-ephys-python-tools`](https://github.com/open-ephys/open-ephys-python-tools/blob/29f869ae4256738e5c74cb55899e182034168710/src/open_ephys/analysis/formats/BinaryRecording.py#L306-L365) package includes a function for writing the necessary metadata file.

jsiegle, 2025-06-08:
The GUI no longer supports the Kwik format, so closing this issue.

closed 2025-06-08 by jsiegle