open-ephys/plugin-GUI #542: Recording failed to start (buffer overload) when using spike detector for two Neuropixels probes

YuelinS, 2022-12-14:
Hi, I’m using Open-ephys v0.6.1 to record from two Neuropixels probes together. I also need online spike detection so am using the spike detector plugin. With two probes, I wasn't able to start the recording because of this buffer error:

[image]

In the case of one probe, the continuous channel buffer monitor will go through an initial peak, then drop to a steady low level afterward. With two probes, however, the peak will go beyond the limit thus causing the recording to stop.

This looks like a solvable problem because:
  1. The initial peak of the buffer monitor seems to be because the computer is creating new folders for the spike-detecting channels all at once. Seems like there just needs to be some optimization at the beginning for how and when to create all the new folders, the following recording is not the problem.
  2. I’ve tried to record the two probes to different nodes (different hard drives) as suggested. But the spikes actually still got saved to both disks. Solving this bug might also help to make spike detection for more than one Neuropixels possible, although not as elegant.

Could you help to take a look? Thanks in advance!

jsiegle, 2022-12-14:
First off, I'll say that online spike detection with Neuropixels probes is very useful for assessing data quality and/or doing closed-loop stimulation, but it's probably not advisable to save all of the waveforms that are detected. The reason for this is that with dense arrays, each spike will trigger threshold crossing on multiple adjacent channels—sometimes as many as 10 or 12. So you end up saving a ton of information that's redundant with the continuous traces. Offline spike sorting can account for this redundancy by assigning a single template to each spike. Accurate analysis of single-unit activity can only be done using offline sorting, and if you're interested in single-channel multi-unit activity it's straightforward to apply a threshold and re-detect spikes offline. So, as a quick fix I would recommend either turning off spike writing by deactivating the "Record Spikes" toggle button, or using two spike detector modules in succession: one for electrodes you're using for real-time analysis (placed before a record node) and one for all other electrodes (placed after a record node).

Second, this is definitely solvable if we return to the previous method for writing spike data, which had one set of files for each spike "group" (spike detection electrodes with the same number of channels on the same stream). In your case, there would just need to be two directories (one per probe). We may consider switching back to this method in the future if opening many files turns out to be a major bottleneck. Alternatively, as you recommended, we could make it possible to specify which electrodes should be written by each record node, so separate threads can handle the file opening and reduce CPU overload.

YuelinS, 2022-12-15:
Thanks for the information! Yes I’m using the spike detector for a closed-loop experiment. I don’t need waveforms indeed but spike timestamps are still needed, so I cannot deactivate the "Record Spikes" toggle altogether.  Actually even if I deactivate spike recording, as long as I place the record node after the spike detector, it will still create spike folders and files for all channels in the beginning, leading to the same buffer overload error when the channel number is larger than 384.

The second method you suggests is our current compromise, which is to only record from a subset of channels, but this is not really ideal for our experiment.

A more compact way of writing spike data would be really helpful to people with similar needs. As apparently the software is fully capable of supporting online spike detection for more than one Neuropixels probe, but is only restricted by the initial file opening part. It will be really awesome if this is something you are willing to implement.

jsiegle, 2022-12-15:
If you only need the spike times, you could use the [Event Broadcaster](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Event-Broadcaster.html) plugin to stream the spike information to Python or Matlab, and write the electrode IDs and timestamps from there.

In the longer term, I think we will likely switch back to the previous directory structure for storing spikes, to prevent the need for numerous files to be open at the same time.

YuelinS, 2023-06-26:
We've tried out the Event Broadcaster and it worked great, thank you!

closed 2023-06-26 by YuelinS