open-ephys/plugin-GUI #543: plugin-GUI no ALL_BUILD file found error

rahmilee, 2022-12-21:
Hello,

I am trying to install the ripple detector file on OE GUI following the protocol from bmsousa91:
https://github.com/bmsousa91/RippleDetector/blob/master/How%20to%20install%20OE%20and%20RippleDetector%20from%20source%20code%20-%20Win10_VS.pdf

When I click the debug file, I get the error that the the ALL_BUILD file does not exist and thus cannot complete the build.

I have tried to re-download the plugin-GUI main file again and re-generate the solution file but not sure how to fix this.
The build file includes the all_build protocol but not a folder in the debug (downloaded from main file). 
[image]
[image]

I have attached some photos below.

rahmilee, 2022-12-21:
I have found the all_build folder with the correct pathway.

I do not understand why the error is showing up that it cannot be found.

Thank you very much!
[image]

jsiegle, 2022-12-22:
Hi there – I'm not sure what could explain this problem, but the instructions in that repo do not match the recommended way to install plugins.

We actually have an updated version of the Ripple Dectector plugin that works with the latest version of the GUI: https://github.com/open-ephys-plugins/ripple-detector

You can follow the instructions [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html) to install the GUI, and [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-plugins.html) to install the plugin.

Be sure your source code directory looks like this before trying to compile the plugin:
```
Code
├── plugin-GUI
│   ├── Build
│   ├── Source
│   └── ...
├── OEPlugins
│   └── ripple-detector
│       ├── Build
│       ├── Source
│       └── ...
```

Also, we are planning to make the Ripple Detector available via the Plugin Installer in early January, if you can wait until then.

rahmilee, 2022-12-23:
Hi Josh,

Thank you very much for the info above.
I will try the protocol you have sent us.

I was wondering if there was a way to the detector can differentiate rem and non-rem sleep?

Thank you very much!
Rahmi Lee

** Caution: EXTERNAL Sender **

jsiegle, 2022-12-23:
I'm not sure about the best way to distinguish REM vs. non-REM. Maybe @bmsousa91 can answer that?

bmsousa91, 2022-12-25:
Hello everyone!

@rahmilee, nice to know that you are interested in using the plugin and I hope it can be useful for you. It does not differentiate between REM and NREM sleep automatically. What we tried to show in the paper is how the sensitivity of the detection can be affected depending on when (which brain state) you run the calibration. For that, we ran an offline sleep classification and then compared the performance of the plugin after being calibrated at distinct moments.

To distinguish between REM and NREM, many authors use the hippocampal theta/delta ratio combined with the EMG signal. Actually, although this is not the main purpose of the plugin, we are currently testing if it is possible to use multiple instances of the plugin to identify REM sleep based on these signals. We tell you when we have the results. Another option is to try to incorporate a classification algorithm into the plugin's structure (beware of the effects on the detection latency). Let me know if I addressed your questions properly!

@jsiegle and team, thanks a lot for updating the plugin to be compatible with the latest version of the Open Ephys GUI!

Best,
Bruno

rahmilee, 2023-02-01:
Hello,

I just wanted to follow up on the ripple detector being added as a plugin. Previously I was told early January it would be uploaded but I have not been able to see it.

Thank you very much!
Rahmi

anjaldoshi, 2023-02-01:
Hi Rahmi,

We just released the plugin via Plugin Installer yesterday, so it should be available to install now. Do let me know if you're not able to find the plugin in the list.

bmsousa91, 2023-04-11:
Hello @AuroraWong, in the GUI you should go to **File > Plugin Installer** and scroll down until you find the Ripple Detector there. Then you click Install. Haven't you found it there?

closed 2023-06-20 by rahmilee