open-ephys/plugin-GUI #544: Go 80 errors for ripple detector plugin

szarrabian, 2022-12-28:
Hello everyone. 
I did as the PDF file indicated to add the ripple detector to the GUI but received 80 errors. can anyone help me please?
[image]

jsiegle, 2023-01-03:
Hi @szarrabian...the instructions in that PDF are a bit outdated.

We actually have an updated version of the Ripple Dectector plugin that works with the latest version of the GUI: https://github.com/open-ephys-plugins/ripple-detector

You can follow the instructions [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html) to install the GUI, and [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-plugins.html) to install the plugin.

Be sure your source code directory looks like this before trying to compile the plugin:
```
Code
├── plugin-GUI
│   ├── Build
│   ├── Source
│   └── ...
├── OEPlugins
│   └── ripple-detector
│       ├── Build
│       ├── Source
│       └── ...
```

Also, we are planning to make the Ripple Detector available via the Plugin Installer by the end of the month, if you can wait until then.

szarrabian, 2023-01-04:
Dear Josh
I appreciate your response.

Best,

*Shahram Zarrabian (Ph.D, Mr)* Assistant Professor,

Department of Anatomical Sciences & Cognitive Neuroscience,

Faculty of Medicine,Tehran Medical Sciences, Islamic Azad University,
Tehran, Iran

A.Email: ***@***.***   |  P.Email: ***@***.***

closed 2025-06-08 by jsiegle