open-ephys/plugin-GUI #545: record node apparently changes the recording path, but stores the data into the old path.

Aryo-Zare, 2023-01-10:
Happy new year
I'm posting this issue for the GUI version 0.6.0 :
The problem is that after once starting & stopping a recording, for the second recording trial; if I want to change the recording directory, apparently I can open the path dialog box & change it, but after restarting the recording, it stores the data into the old path !
I made a work–around by deleting the record node & again drag & dropping it from the Processor List. When doing this the newly changed path would work for 1 record start-stop. For the next recording trial, this process should be repeated.

jsiegle, 2023-01-11:
To have the new path take effect, you just need to press the "plus" button in the control panel to start a new recording directory. Otherwise, it will continue to save data in the original location.

We could modify the behavior so that updating the path of any Record Node automatically creates a new directory. But since this change would apply to all Record Nodes, I think it's better to require the user to manually press the new directory button.

alekhka, 2023-12-20:
Is there a way to press the "plus" button via remote control?

jsiegle, 2023-12-21:
Yes, just sent a `PUT` request to `http://localhost:37497/api/recording` with the following contents:

```json
{"start_new_directory" : "true"}
```

In Python, this would be:

```python
import requests
r = requests.put(
    "http://localhost:37497/api/recording",
    json={"start_new_directory" : "true"})
```

This is not currently in the documentation, but we'll get it added!

medengineer, 2025-06-09:
This feature has been implemented in v1.0 in [this](https://github.com/open-ephys/plugin-GUI/commit/82b97a42e76a522706aa6971e9fc8032e0d0d79e) commit. Details can be found in the updated "Recording data" section once v1.0 is officially released.

In short, a new "force" button with the lock symbol has been added to the right of the "new directory" button. Selecting this button will effectively lock the "new directory" button such that it stays enabled, creating new directories between each recording for all Record Nodes.

closed 2025-06-09 by medengineer