open-ephys/plugin-GUI #547: Custom textEvent in plugin

ANickerson, 2023-01-19:
Hello,

I am trying to port a plugin from 0.5.x to the latest version 0.6.3.
I previously had a custom text event to log a json object, which was working well.

In the new version I cannot get it to work.
I add the event channel in the "updateSettings" section of a "GenericProcessor".
But when i use `addEvent(e)` it crashes on deserialization.

looking at GenericProcessor line 1245:
```cpp
            {

                TextEventPtr textEvent = TextEvent::deserialize(dataptr, getMessageChannel());

                handleBroadcastMessage(textEvent->getText());
            }

```
this looks like its trying to deserialise with the getMessageChannel(), rather than the new event channel.

Can you advise how to add a custom text event channel in the new version?

(p.s. thanks for your work here! its awesome)

jsiegle, 2023-01-19:
Hi Aiden – in v0.6.x, every plugin automatically has access to a message channel that is used to send and receive text events. If you just call `broadcastMessage()` inside the `process()` method with your JSON string as the input argument, it will send a message to all other plugins, which will also get saved by any Record Nodes in the signal chain. The timing will be accurate to within one processing callback, so if you need a precise timestamp you should include that in your message.

Does that address the issue?

ANickerson, 2023-01-22:
Hi,
That works as workaround solution, yes thanks!

However currently my downstream analyses are looking for the channel "protocol.event" (the channel name) to easily split the events from other information.
What would you recommend in this case (to easily find these events).

jsiegle, 2023-01-22:
Can you just prepend the messages with the same string (e.g. `protocol.event`)? Then you can easily filter the values that show up in the MessageCenter `text.npy` file.