open-ephys/plugin-GUI #548: Digital input sample numbers not matching binary files

DoktorDu, 2023-01-27:
Hello, 

I have noticed that when recording in binary mode the Event timestamps/sample numbers from digital inputs (in files timestamps.npy and sample_numbers.npy) do not always correspond to the timestamps in binary files. Specifically, if writing to disk is paused at any point but acquisition is still ongoing, the Event sample numbers reflect total samples acquired so far rather than total samples written to disc. 

What follows is that if the experimental session consists of multiple Recordings within a single Experiment folder it is not possible to match the Event sample numbers/timestamps to the binary file because the periods when data was being acquired but not written to disc are not accounted for.  For example, if the recording is started 20 seconds after acquisition and an Event (digital input) happens at the same time, it will have a 20sec timestamp in the timestamps.npy file but will correspond to the beginning of the binary file, with no way of knowing the offset between the two post-hoc. Is there a a way around it? For example, is there a way to record Even sample numbers relative to individual binary files (i.e. starting from 1 every time record button is pressed and new recording folder is created)? 

I have only noticed this problem after a few recording sessions, I don't suppose there is a way to align the Event timestamps with individual binary files post-hoc? 

Thank you in advance for any advice!

medengineer, 2023-01-27:
All timestamps are relative to the start of acquisition. At the start of recording, the current timestamp for each stream is logged in a sync_timestamps.txt file found at the same level as the continuous and events folders within a Binary recording. 

If you want to align event timestamps to the raw continuous data, you can subtract the start timestamp for the corresponding stream the event belongs to in the sync_messages.txt file. Equivalently, you can subtract the first timestamp found in the sample_numbers.npy file for the corresponding stream.

chris-angeloni, 2024-05-21:
Following up on this, is it expected to have some negative LFP samples relative to the start of a recording?

For instance, if I am looking at my continuous data: the first timestamp of the LFP data is 22.7446s

medengineer, 2024-05-22:
To clarify, sample numbers represent the number of samples received since the start of acquisition within a _single_ stream. Timestamps (in seconds) are computed using the sample numbers of sync events across _multiple_ streams as the sync events are detected by the Record Node. 

Depending on which stream is designated main vs. auxiliary in the Record Node, it’s possible for computed timestamps to appear negative within a stream _locally_. The timestamp values should instead be considered _globally_ when aligning data from the auxiliary streams to the main stream during offline analysis.

In fact, there are multiple reasons why the values in `timestamps.npy` may not be equivalent to the values in `sample_numbers.npy` divided by the sampling rate, including differences between the theoretical and actual sample rate of each device, as well as the relative timing of sync pulses across streams.

chris-angeloni, 2024-05-22:
Hmm, I see. So if the recording node was performing online synchronization, how would I compare the times in `timestamps.npy` to say the spike times returned from kilosort (sample number in the recording). It sounds like `spike_times / sample_rate` should be in the same time base as `timestamps.npy` minus the first timestamp?

medengineer, 2024-05-22:
Each kilosort spike sample number maps to an index in `sample_numbers.npy`. You can use those same indices to extract the corresponding timestamp values from `timestamps.npy`

jsiegle, 2024-05-22:
It's actually simpler than that – the sample numbers that are output by Kilosort can be used directly as indices into `timestamps.npy` to map each spike time to a global time in seconds.