open-ephys/plugin-GUI #553: Upgrade from v0.5.5.4->0.6.3, timestamps in events files now in sample #, instead of time in sec.

jays314159, 2023-02-10:
When I load an events file in Matlab using 'load_open_ephys_data' to read TTL events and their times, the timestamps I got before used to be in seconds, matching that of times in '.continuous' files. After the upgrade, they seem to be measured as sample numbers. I'm using Intan.

Is this an intentional change? I wasn't sure in light of version update doc. that said it's now using consistent naming and numbering conventions.

medengineer, 2023-02-10:
The change is intentional, as of v0.6.0 we wanted to remove the ambiguity between sample_numbers and timestamps across formats. 

I recommend checking out the [open-ephys-matlab-tools repository](https://github.com/open-ephys/open-ephys-matlab-tools) that provides a more user-friendly object-oriented interface that is up-to-date with the latest changes. The examples folder shows how timestamps / sample_numbers are used and consistent across all formats.

jays314159, 2023-02-11:
Ah, OK. 

I upgraded to use the native HTTP server functionality, but because we already have pipeline setup for the old version (and don't feel like changing it yet again), I'll have to go back to previous version and use Network Events plugin. 

Thank you for quick confirmation, in any case.

closed 2023-02-11 by jays314159