open-ephys/plugin-GUI #555: Notch Filter

PostDocKala, 2023-02-14:
Hi, I was wondering how one can enable the notch filter on the Open Ephys GUI? I can't seem to find the setting anywhere.

medengineer, 2023-02-15:
Hi, 

There is no notch filter in the most recent versions of the GUI. If you are trying to remove noise from a narrow frequency band (eg 60Hz interference) we recommend using the Common Average Reference plugin for this instead.

stijnphx, 2023-10-18:
Hi,

How should you use it? Say there is a 60 Hz interference on all channels, then I should select all channels in reference channels and in affected channels, right?

Thanks.

jsiegle, 2023-10-18:
That's right...assuming the 60 Hz noise occurs on all channels, they should all be selected as both "reference" and "affected."

jsiegle, 2024-05-27:
We're not planning to add a Notch Filter plugin. Are you seeing 60 Hz noise in your data that's not removed by the Common Avg Ref?

florian6973, 2024-05-30:
Yes indeed, since some electrodes have different impedances, the CAR does not remove the noise well.

Moreover, when disabling some channels in the Channel Map (located before the CAR component in the signal chain), OpenEphys crashes when restarting the recording, if I do not manually update the Reference channels in the CAR. Is it expected please? (see log 
[activity_2024-05-29_08-35-07.log](https://github.com/open-ephys/plugin-GUI/files/15492928/activity_2024-05-29_08-35-07.log))

anjaldoshi, 2024-06-04:
> Yes indeed, since some electrodes have different impedances, the CAR does not remove the noise well.

You can handle the case of mismatched impedances by using multiple CAR plugins for each set of electrodes.

> Moreover, when disabling some channels in the Channel Map (located before the CAR component in the signal chain), OpenEphys crashes when restarting the recording, if I do not manually update the Reference channels in the CAR. Is it expected please?

Thanks for bringing this to our attention. We were able to repro this crash on our end. As a quick fix, I patched the CAR plugin and generated a new DLL for it. You can update the plugin on your end locally by simply extracting the below zip file and then replace the plugin DLL located in the plugins directory next to you GUI executable with the new one. If you installed the GUI at the default location, the plugins folder can be found here `C:\Program Files\Open Ephys\plugins`

[CommonAverageRef.zip](https://github.com/user-attachments/files/15568976/CommonAverageRef.zip)

florian6973, 2024-06-09:
Thank you so much for your prompt fix, it works like a charm now!

Indeed, however given the number of different electrodes we have, this is not very practical... What would be the difficulties to implement a Notch Filter by ourselves please, as a plugin?

jsiegle, 2024-06-10:
This would actually be pretty straightforward – you could copy the code for the FilterNode plugin and change the filter type from `BandPass` to `BandStop`: https://github.com/open-ephys/plugin-GUI/blob/8c6453954b431a1f48f18b8fc3ddd1282eeda56d/Plugins/FilterNode/FilterNode.cpp#L38

Instructions for creating a new plugin can be found [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Creating-a-new-plugin.html).

florian6973, 2024-06-11:
Thanks for your feedback! I just have an issue regarding the symbols. I am using VS 2022 Community Edition to compile the plugin, but I cannot use the existing open ephys plugins with my compiled version:
[image]
Reciprocally, if I copy the generated DLL to my installed OE version, OE cannot load the custom DLL with a similar error. How could I solve it please?
The repo is https://github.com/florian6973/notch-filter-plugin

Thanks!

florian6973, 2024-06-11:
Thanks for your help! Here is the repo https://github.com/florian6973/notch-filter-plugin
Clearly the longest part was to setup compilation... Would you like it to distribute it or not please?

florian6973, 2024-06-11:
Here is what it looks like: https://drive.google.com/file/d/1d_4tWGzD4yRvkzVA_71-nD4zC20koLwT/view?usp=sharing

jsiegle, 2024-06-16:
Thanks for putting this together! I think it would be reasonable for us to release this, with the caveat that it should only be used if other methods of reducing noise fail.

I just created a forl so we can distribute it via the Plugin Installer: https://github.com/open-ephys-plugins/notch-filter

florian6973, 2024-06-20:
Fantastic, thank you very much! So it should be available in the plugin installer tool of the next OE version?

jsiegle, 2024-06-21:
Plugins are released independently of the core application, so it will show up in the plugin installer once we set it up to build automatically. We'll let you know when it's ready!