open-ephys/plugin-GUI #557: v0.5.5.4 impedance measurement not saved with Rhythm FPGA

aldenshoup, 2023-03-02:
When I make an impedance measurement it does not save anywhere despite the file path being set and the "save impedance measurement" option being checked.

jsiegle, 2023-03-03:
This is a bug in v0.5.x, which has been fixed in the latest version (v0.6.x). If possible, I'd recommend upgrading to take advantage of the many usability and performance improvements in v0.6.x. If you can't fully transition to the latest version at this point, then you can use a [.zip file installation](https://github.com/open-ephys/plugin-GUI/releases/tag/v0.6.3) to run v0.6.x side-by-side with your existing version, use the Acquisition Board plugin to measure and save impedances, and then run your experiment with v0.5.x.

aldenshoup, 2023-03-26:
Great thank you so much! It's working just fine on v0.6.x.

aldenshoup, 2023-03-26:
I actually have an additional question on this topic, I was wondering what frequency you select for the Intan RHD2000-series impedance measurement? In v0.5x the console would output something along the lines of "switching to 0.1 pF capacitance" or something like that but I wanted to know more about the impedance measurement process on your software. Thanks so much!

jsiegle, 2023-03-27:
The measurement frequency is still [1000 Hz](https://github.com/open-ephys-plugins/rhythm-plugins/blob/c416a1341879a5b9b3c690f768d446d971c0abf6/Source/ImpedanceMeter.cpp#L315). All measurement parameters are the same in the latest version, there's just less info printed to the console.

closed 2025-06-08 by jsiegle