open-ephys/plugin-GUI #558: xcode and downloading plugins

jonahpc, 2023-03-08:
I have followed the instructions of "https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491555/macOS" and when I open xcode I get over 11 errors and 5 warning. I need help figuring out how to download these plugins! What am I missing?
[image]

medengineer, 2023-03-08:
Hi,

Are you building the development branch? We recently bumped the min OSX version to 10.15 on there but the main branch should still work with 10.9.

anjaldoshi, 2023-03-08:
Also, what version of MacOS are your running? And, as mentioned on the top of the wiki page you linked, we have moved all the documentation for the GUI here: https://open-ephys.github.io/gui-docs/ . Up-to-date instructions for compiling the GUI from source can be found [here](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html).

jonahpc, 2023-03-08:
I think the main issue is the plug in downloads that I want to access. It
keeps saying this:

jsiegle, 2023-03-09:
Hi Jonah – it doesn't look like the error message came through. 

Which plugins do you want to use? If running the GUI via Xcode, you won't be able to use plugins downloaded via the Plugin Installer, as these are built in "Release" mode.

closed 2025-06-08 by jsiegle