open-ephys/plugin-GUI #561: Ability to send 8 neural channels to OE acq board analog out port for use with I/O board

ChucklesOnGitHub, 2023-03-22:
We've had a user request make channel selection available to output 8 analog headstage signals via the analog out port of the OE acquistion board. They could then be used downstream by connecting an I/O board.

It would extend the current functionality of the Audio Out in the rhythm plugins that outputs two analog channels to the board's audio jack.

jsiegle, 2023-03-23:
@aacuevas does this require any changes to the firmware? If not, we can add it in the next release.

aacuevas, 2023-03-24:
No, we already use the API to redirect channels to outputs 1 and 2, which are also routed to the jack. We just lack a UI to set the output to all other 6 channels.

paulodecastroaguiar, 2023-03-26:
@aacuevas is there code or a specific plugin that I could try to adapt/modify to create this UI?
(thank you for your help on this!; I was the one originating this user request)

jsiegle, 2023-03-26:
[These are the buttons](https://github.com/open-ephys-plugins/rhythm-plugins/blob/c416a1341879a5b9b3c690f768d446d971c0abf6/Source/DeviceEditor.cpp#L448-L456) that are used to select the channels to map to DAC1 and DAC2. You could create additional buttons for DAC3-8 or, alternatively, just have one button that displays a [PopupChannelSelector](https://github.com/open-ephys/plugin-GUI/blob/5706680e8dbacafde0bb43a2839c5ae897d0646b/Source/Processors/Editors/PopupChannelSelector.h) with a [maximum of 8 selectable channels](https://github.com/open-ephys/plugin-GUI/blob/5706680e8dbacafde0bb43a2839c5ae897d0646b/Source/Processors/Editors/PopupChannelSelector.h#L140).