open-ephys/plugin-GUI #562: New sub-directory for recording - via remote control

larsrollik, 2023-03-31:
After upgrading from 0.5.x to 0.6.4, the network control does not allow anymore the use of a new sub-directory that does not exist yet. This used to work, but in the new version the GUI "stops working" with a stack trace of thread errors.

Example message used:
`StartRecord CreateNewDir=1 RecDir=EXISTING_PATH\NEW_SESSION_DIR PrependText=SESSION_NAME AppendText= `

The GUI runs on a Windows 10 machine.

I also tried to replicate with the new HTTP server and run into the same issue, whenever I set the `parent_directory` to a non-existent sub-directory.

Was this only working by accident before or is there a way to create this folder structure with the current GUI version?

Thank you.

larsrollik, 2023-06-09:
Sorry for the delay.
There is actually no stack trace or log entry about this. After confirming a dialog message "open ephys has stopped working", the main GUI closes. No further log is written to the text file or console.

Can I ask why the subdirs are not created in the HTTP approach?
From glancing over the path handling in the GUI controller, it seemed that it would only need to make the missing directories as you would in `mkdir -p` for example. Or are the paths handled differently?

Are you suggesting to create the new acquisition via ssh prior to acquisition? 
I am interested in using the HTTPServer approach because it gets around fiddling with remote access to Windows machines.

medengineer, 2023-06-23:
I just tested setting non-existent parent directories via HTTPServer on both Mac and Windows and the sub dirs are getting created accordingly. 

This is with GUI v0.6.5 and python tools v0.1.5.

I wrote an example script [here](https://github.com/medengineer/open-ephys-test-suite/commit/3a75fdf6a9d559860d49712bae1896c68e16d657).

larsrollik, 2023-06-23:
Great, thank you! I will test the updated GUI version.

closed 2025-06-08 by jsiegle