open-ephys/plugin-GUI #563: Spike detector - stream waveforms of all channels when detecting a spike

MarineChap, 2023-03-31:
Dear, 

Would it be possible to stream the waveforms or eventually just the amplitude of all channels when detecting a spike ? 
From what I can see, I can only have the waveform from channels grouped by 1, 2 or 4 channels but I need the amplitude at spike time of all (my 8) channels. My  guess is that you did not implement it already because this would not work for high-density probes or is there another reason ? 

Thank you.
Have a good day

jsiegle, 2023-03-31:
To do this properly, you'd need to make it possible for the Spike Detector to work with "octrodes" (a set of 8 channels that are sampled simultaneously). This is definitely doable, but it would take a bit of work.

A simpler approach would be to set up 8 single electrodes on each channel and stream the amplitudes using the [Event Broadcaster](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Event-Broadcaster.html) (the JSON packet includes a field for the amplitude of each spike). Any electrodes without near-simultaneous spikes can be assumed to have sub-threshold amplitudes. Obviously this doesn't give you complete information about the spike, but it may be sufficient for some applications.

MarineChap, 2023-03-31:
If I try to implement it, would you be interested to merge it  at some point? 

About the simplest approach you described, not sure how EventBroadcaster can help as I want to keep using the data inside open-ephys for this project. It is a sink no? So I cannot use the json output inside Open-ephys right? 

Other idea, can you have a plugin with two input datastreams ? Like a plugin that take as input the output of spike detector as well as the continuous signal. Then in my plugin I can, based on the spike timestamp, select in the continuous signal the amplitudes I am looking for. 
I know in Falcon it is possible even if you need to be careful of packet synchronization between the two inputs. Is that possible with open-ephys ? 

Thanks.

jsiegle, 2023-03-31:
Thanks, I think I understand the application better now.

The continuous signals pass through the Spike Detector unchanged, so the next plugin down the line could take the spike times and extract the full continuous signal +/- 4 channels away. The spike sample numbers are aligned with the continuous data at the time of detection, so synchronization will not be an issue. I think this is preferable to creating an octrode in the Spike Detector, because in this case you wouldn't have to pre-define which channels are sampled simultaneously, and can instead sample the channels around the largest-amplitude waveform.

MarineChap, 2023-04-04:
Ok that's clear. It is actually way easier than I was expected. I won't develop this part before a month. I think we can close this issue and eventually if it does not work as expected for some reason, I can come back on this issue later. 

Thank you for your input !

MarineChap, 2023-05-16:
Hello, Sorry I need some help with what we said before.  The amplitudes found are zero and nan until the app crashed. 
I am not sure with the new API that I am using correctly the sample id and the channel number correctly.
Do you see what I am doing wrong ? 
Thanks.

```
void ProcessorPlugin::process(AudioBuffer<float>& buffer)
{
    // save continuous buffer in variable
    dataBuffer = &buffer;
    checkForEvents(true);

}

void ProcessorPlugin::handleSpike(SpikePtr spike)
{
    //  Extract from continuous data buffer the amplitude data for each channel at each spike time.

    for (int chan = 0; chan < dataBuffer->getNumChannels(); ++chan)
    {
        float amplitude = *dataBuffer->getReadPointer(spike->spikeChannel->globalChannelIndexes[chan], 
                                                                    spike->getSampleNumber()));
    }
}
```

anjaldoshi, 2023-05-16:
Hi Marine,

Looks like there are a couple of things wrong with the way you're getting the amplitude of each channel at the spike's sample number. 

First of all, the channel number passed in the `getReadPointer()` call should be just `chan` as you just want to extract amplitude from each channel. Second, the sample number obtained from `spike->getSampleNumber()` is relative to the start of the acquisition, but the sample number here should be relative to the start of the current buffer. So, you need to subtract the first sample number of the buffer  from `spike->getSampleNumber()` to get the spike's sample number relative to the buffer. The first sample number of the buffer for a stream can be obtained by calling `getFirstSampleNumberForBlock()`.

anjaldoshi, 2023-06-13:
We looked into the Spike Detector code to understand what was going on and realized that  negative sample index is actually expected, due to the overflow buffer in the spike detector: https://github.com/open-ephys/plugin-GUI/blob/7961480a3fe47c3db35be8176300a10293b6d2b0/Plugins/BasicSpikeDisplay/SpikeDetector/SpikeDetector.cpp#L748-L771

At the end of each buffer 200 samples are stored in case a spike occurs at the edge. If this wasn't done, there would be spike waveforms that get cut off in the middle. Spikes detected at the beginning of the next buffer will have sample numbers before the start of the buffer.

To deal with this, you would need to implement a buffering system similar to Spike Detector's in your plugin. It should be possible to just copy what's done in the Spike Detector with some minor changes.

Let us know if you need any further help.

MarineChap, 2023-06-15:
Ok thank for the debugging. 

I think I fixed it by saving the previous buffer in an overflow buffer and then finding the index that way. 

```
        auto index = spike->getSampleNumber() - getFirstSampleNumberForBlock(spike->getStreamId());
        if( index > 0){
            spikePositionParticle.push_back(*dataBuffer->getReadPointer(chan, index));
        }else if(overflowBuffer != nullptr){
            spikePositionParticle.push_back(*overflowBuffer->getReadPointer(chan, overflowBuffer->getNumSamples() + index));
        }
        else{
            LOGD("Spike not found back in the dataset.");
            return;
        }
```
Does this sound correct for you ? At least I don't have error anymore but I am not sure how I could easily check that I have indeed the right timestamps.

jsiegle, 2023-06-15:
That implementation looks good. To check that all the spikes are being handled, you could add counters that print the total number of spikes packaged by the Spike Detector and received by your plugin after acquisition ends.

That's a good point about the buffer size. If the GUI's internal buffer is less than 200 samples, the spike detector won't work properly. I just added a warning about this [in the docs](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Spike-Detector.html#spike-detector).

It's not possible to add metadata to events after they are created. Instead, you should create new spike events with the same timestamps and add metadata to those. Your plugin can just create one `SpikeChannel` in the `updateSettings` method, and use this to broadcast a new set of spikes to downstream plugins.

closed 2023-07-03 by MarineChap