open-ephys/plugin-GUI #568: Problem installing

KEStanchak, 2023-04-18:
Hi -- We're new to using Open Ephys, and we're having a problem connecting to the Acquisition Board on macOS. We get the same issue on both 10.15.7 (Catalina) and 11.0.1 (Big Sur). 

When we try to install the plugin OE FPGA Acquisition Board, the source does not load in the GUI, and we get an error "Unable to load OE FPGA Acquisition Board in the Processor List". 

We noticed that when installing Open Ephys in Windows 11, there was an option to install the driver: "Opal Kelly Front Panel USB Driver for Open Ephys Acquisition Board". Is this or the mac equivalent missing from the mac installation? 

Thanks for any help!

jsiegle, 2023-04-18:
Hi Katie – I assume you have a board that was recently purchased from the Open Ephys store (last 6 months)? In that case you're using the right plugin, but there's a separate step required for installing the driver on Mac. Can you follow the instructions [here](https://open-ephys.github.io/acq-board-docs/User-Manual/New-FPGA.html#on-macos) and let me know if that fixes the issue?

jsiegle, 2023-04-19:
Actually, we just found a problem with the Mac binaries that’s causing this. We will put out an updated release tomorrow that should fix the issue.

KEStanchak, 2023-04-19:
Thanks so much! We'll keep an eye out for it.

jsiegle, 2023-04-19:
Ok, the new release is up. If you use the plugin installer to downgrade to an earlier version, then upgrade to the latest version (0.2.0), it will download the repaired Mac binaries. You also shouldn't need to install the driver separately, since that is now included along with the plugin installer download.

Let me know if that works!

KEStanchak, 2023-04-21:
Hi Josh,

Thanks for all of your help with this. Unfortunately, we're still having the same problem. I deleted then reinstalled the open ephys gui in my Applications folder, but I'm still unable to get plugins to load in the Processor list. I tried OE FPGA Acquisition Board, Rhythms Plugins, and Spike Sorter. I also tried downgrading the plugins and them upgrading, as well as uninstalling and reinstalling the plugins.

anjaldoshi, 2023-04-21:
Hi Katie - Can you tell us what version of the GUI are you using? 

Also, from what you're saying, you're not able to get any plugins to load that are plugins installed via Plugin Installer. Can you try deleting everything from the `$HOME/Library/Application Support/open-ephys`  folder, then launch the GUI and install any of the plugins via Plugin Installer again? The user Library folder is hidden by default in macOS and can be accessed by following the instructions [here](https://helpx.adobe.com/x-productkb/global/access-hidden-user-library-files.html).

KEStanchak, 2023-04-21:
I'm using 0.6.4. I deleted the three folders from the open ephys application support folder, reopened the GUI, attempted to install OE FPGA Acquisition Board and Spike Sorter via the Plugin Installer, and in each case it was "installed" (i.e., showed up in the "installed" list of the Plugin installer) but a message pops up saying "Unable to load in the Processor List". This was on Catalina.

I just tried installing 0.6.4 on a machine running Big Sur. The Plugin Installer shows that "Rhythm Plugins" and "OE FPGA Acquisition Board" are installed, but they do not show up in the Processor List. (IIRC, this was the case this morning when I reinstalled 0.6.4 on the Catalina machine.) When I try uninstalling and then reinstalling those plugins, I get the same message "Unable to load in the Processor List"

Thanks, and I'm sorry for all of the trouble!

anjaldoshi, 2023-04-22:
Thanks for trying different things out and giving us more details. This is strange, especially since it is not working on two different machines. One more thing I wanted to confirm was you're using the .dmg installer to install the GUI on macOS, right? 
Also, can you try installing the Spike Sorter plugin manually? I'm attaching the plugin zip file here. Just extract it and copy the contents inside the plugins folder to your `$HOME/Library/Application Support/open-ephys/plugins-api8` folder. Then launch the GUI. and see if it shows up in the Processor list.

KEStanchak, 2023-04-27:
It still doesn't show up in the "Processors" list in the GUI. (This is on catalina.) 

We just had a new open ephys box delivered with the new FPGA, so I'm now trying that. Today I reinstalled the open ephys GUI (from the dmg). The Spike Sorter, OE FPGA Acquisition Board, and Open Ephys Data Format plugins were installed (based on the Plugin installer GUI). However, they still did not show up in the Processor list, even after downgrading and upgrading. 

I then tried downloading and copying the libftd3xx.dylib driver to /usr/local/lib folder as stated in the instructions that came in the box (although I see it is in the shared-api8 along with two other drivers). The catalina I'm running did not already have a lib folder within /usr/local (which I thought was weird). I did not disable SIP to copy to /usr/lib. Anyway, that doesn't allow me to see the downloaded plugins in the processor list, either. Then I tried replacing the Spike sorter bundles in the plugins-api8 folder with the bundle from the zip file you linked to. So spike sorter shows up as downloaded from the Plugin Installer, but still doesn't list in the GUI.

KEStanchak, 2023-04-27:
Sorry for all of the trouble! I just wanted to add that I just now went to the Big Sur machine, deleted the open-ephys app and application support folder, redownloaded open-ephys using the dmg, and was still unable to get the spike sorter plugin, neither from the plugin installer nor the zipped file you linked to.

anjaldoshi, 2023-04-27:
No trouble at all. This is the first time we're encountering this, and I feel like it is a security / permissions related issue. So, it is a bit hard for us to debug without having access to the actual system that is facing such issues. It'd be great if we can setup a video meeting where you'll share your screen from the machine that is facing this issue and we can try some stuff to debug, or at least pinpoint this issue. You can shoot me an email at anjal@open-ephys.org and we can continue from there.

anjaldoshi, 2025-06-09:
I believe this issue has been resolved in v1.0  with the introduction of codesigned binaries for the GUI and all plugins. Additionally, we’ve unified the Acquisition Board plugin in v1.0, ensuring compatibility across all Acquisition Board versions (Opal Kelly, ONI, and 3rd gen). The plugin now includes a statically linked FTDI driver, which should further reduce the likelihood of this issue occurring.

If you continue to experience this problem in v1.0, feel free to reopen the issue.