open-ephys/plugin-GUI #569: Cannot see spikes

rjp109, 2023-04-28:
Hello,

We've recently conducted recordings using a 64ch silicon probe. When we first started these recordings we were having trouble seeing any spike activity when a bandpass (300-6000Hz) filter was applied on the OE gui. I had a feeling there was some issue with our setup meaning the filter wasn't being applied properly. Oddly when we unticked and re-ticked the 'AUX' button in the acquisition board window and then re-started the LFP viewer, the data looked much better and there seemed to be a lot of spiking activity. The spike detector function was also picking this activity up - so we continued recording and now have a couple of weeks of data.

I'm now in the process of spike sorting this data offline but I am not having much luck in identifying single units. I have since looked back at the raw data on OE and other software with a bandpass filter applied, and I can no longer see this spiking activity.

Does anyone have any idea what is going on here? Why would unticking and ticking the AUX button change the data shown in the LFP viewer?

I'm hoping the spike data is there somewhere, we just need to apply some other kind of post-processing...

jsiegle, 2023-04-30:
Did you happen to take any screenshots of the LFP Viewer during the experiment? I'm curious what the signals looked like.  It would also be helpful to see some plots of the recorded data.

The AUX button should only add/subtract 3 channels per headstage (which will only have meaningful signals if you have accelerometers on your headstages).

rjp109, 2023-05-02:
Hello,

Unfortunately the photos attached are all I have. The top photo is what it looked like before checking and unchecking the AUX button (top is unfiltered, bottom filtered between 300-6000Hz). The photo below is afterwards - admittedly the activity in this second picture isn't super convincing of real spikes but it did look convincing during other times of the recording - either way what the LFP viewer is showing has drastically changed. Would it help to upload a recording? I'd really like to at least be able to recreate what we were seeing on the OE gui 
[image]
[image]
[image]

jsiegle, 2023-05-05:
What type of electrodes are you using / what is the electrode layout? It's strange that the bandpass-filtered data on the bottom changed after toggling the AUX button, but the data on top did not (and doesn't show any spikes).

rjp109, 2023-05-05:
This is a Neuronexus 8 shank silicon probe with 8 electrodes per shank and each shank 200um apart. 

The gui setup had a splitter with two LFP viewers, so the top LFP viewer was filtered between 0.5-30Hz and the bottom 300-6000Hz. The top/bottom photo before/after checking the aux button.
But yes really bizarre the difference the AUX button made to the high-pass filtered data. I originally thought it was not the AUX button itself but more just changing the settings which resets application of the filter, but as i said, i coud not recreate the second image with the spikes when I opened the data in the gui offline.

jsiegle, 2023-05-06:
I'm sorry to say I'm not sure what's going on here. If the spikes showed up in the LFP Viewer / Spike Detector, they should have been saved by the Record Node.

We can take a look at a segment of the recording if you want to upload it.

jsiegle, 2025-06-08:
It's not clear what the problem is -- if it persists please re-open this issue.

closed 2025-06-08 by jsiegle