open-ephys/plugin-GUI #570: Base install should come with support for non Opal-Kelly based acquisition boards

jonnew, 2023-05-12:
The "OE FPGA Acquisition Board" Plugin should be available by default in the base installation since all new acquisition boards uses this module.

jsiegle, 2023-05-14:
The latest `development` branch [now asks the user which type of acquisition board they have](https://github.com/open-ephys/plugin-GUI/blob/8d7dd8ac30989e7eddec50164ec9f66ab8030486/Source/UI/DefaultConfig.cpp#L222-L227) before loading a default configuration. We will put out an update with that change this week.

Longer term, the acquisition board plugin will be able to auto-detect which type of board is plugged in, so this won't be an issue.

Including externally hosted plugins in the base installation is not currently possible, but we can add this ability in the next major release.

jonnew, 2023-05-22:
Nice, this is a good solution. I guess we can leave this open and close it when the auto-detect feature is ready?

jsiegle, 2023-05-22:
Yep, that sounds good. We are planning to put out the new release of the core application today.