open-ephys/plugin-GUI #571: Spike Sorter shows only 1 active Tetrode

ShaynDoe, 2023-05-12:
Hello, I recently tried using the Spike Sorter plugin on the OpenEphys GUI (v0.6.2) for my data (acquired with tetrodes). Somehow the latest version of the plugin (0.2.0) only shows 1 active tetrode - tried this with both File Reader and Acquisition Board as the source. The downgraded version of the plugin (0.1.0) doesn't have this problem.

medengineer, 2023-05-12:
Hi, can you attach the signal chain that uses FileReader and causes this issue?

ShaynDoe, 2023-05-22:
Hello, sorry for the delayed response. Please find the signal chain attached, and as can be seen - only tetrode 1 is active (this is with the 0.2.0 version of Spike Sorter). 

[image]

medengineer, 2023-05-22:
Looks like Spike Sorter 0.2.0 requires GUI v0.6.4 where we updated how electrode settings are managed by the GUI. I'd recommend upgrading to GUI v0.6.4 if possible and your existing configuration will load properly.  

However, if you weren't having issues with the previous Spike Sorter and prefer to stay on GUI v0.6.2, rolling back Spike Sorter to the previous version via PluginInstaller is also an option.

closed 2025-06-08 by jsiegle