open-ephys/plugin-GUI #572: rhd2000_usb3.bit not found using acquisition board

neuronzoo, 2023-06-13:
Hi, 
I am using v0.6.4 with an open ephys acquisition board bought 3-5 yrs ago. I have the "Rhythm Plugins" plugin installed.
The acq board is acquiring some basic TTL data as analog input, and EMG through a TDT headstage (intan 12-pin connector).
This has worked fine previously. Today, upon opening the GUI, I got the following message:
`The rhd2000_usb3.bit file was not found in the directory of the executable. Would you like to browse for it?`
-> I didn't change anything in the hardware
-> The file is in "C:\Program Files\Open Ephys\shared". I tried browsing for it, found it, but message keeps appearing. I tried redownloading from github, didn't help.
-> Changed version to 0.6.5 - same problem.
-> Tried uninstalling and installing the rhythm plugins, didn't help.
-> Tried moving to a different USB(2) slot - didn't help.
Help appreciated!

anjaldoshi, 2023-06-14:
Hi,

Can you try copy-pasting the `rhd2000_usb3.bit` file from the  `C:\Program Files\Open Ephys\shared` directory to the `C:\ProgramData\Open Ephys\shared-api8` directory if it is not there already. Note that `Program Files` and `ProgramData` are two separate directories, and ProgramData is a hidden by default. You can make it visible by following the steps [here](https://support.microsoft.com/en-us/windows/view-hidden-files-and-folders-in-windows-97fbc472-c603-9d90-91d0-1166d1d9f4b5).

jsiegle, 2023-06-15:
Given that the problem appeared randomly without any changes to the software, and persists across multiple versions of the software, I think a hardware failure (defective FPGA) is the most likely cause. Can you please send a message to `support@oeps.tech` (hardware support) with a link to this issue?

Mizrahi-Kliger, 2023-06-20:
Thank you Josh, 
I tried to replace my AC/DC power adapter from (the non-original) 5V 1A output to a 5V 3A one. Message went away and seems like the Acq board works well.

neuronzoo, 2023-06-20:
So, I replaced the AC/DC adapter I had (a non original one) that gave out 5V 1A with a 5V 3A one, and the problem went away!
Thank you guys!

ChucklesOnGitHub, 2023-06-20:
Required input is 5V, 2A

closed 2023-06-20 by ChucklesOnGitHub