open-ephys/plugin-GUI #573: Display issue when opening GUI

ChucklesOnGitHub, 2023-06-15:
[image]

Recurrent user issue where the console log shows everything is running but the GUI is not displayed. The Reloading Settings message  makes ppl think something is amiss.

An easy way to get it back with the GUI open is to: 
1. Hover the mouse over the Open Ephys GUI shortcut that is on the Windows Taskbar until the two miniature windows appear
2. Right-click on the miniature of the main GUI window
3. Select Maximize. 

It would be great to have the window state check screen boundaries to avoid having to do this.

anjaldoshi, 2023-06-15:
Hi Ceci,

We were able to identify this issue on our end, and noticed that this happens when the monitor configuration changes between sessions. I already pushed a fix for this on the dev branch (https://github.com/open-ephys/plugin-GUI/commit/86b6c558fdc919972a5ed28485a8a6d923bbd1ef) a few days ago. It will get merged in the next release. 

-Anjal

anjaldoshi, 2023-08-17:
Fixed in version 0.6.6.

closed 2023-08-17 by anjaldoshi