open-ephys/plugin-GUI #574: Electrode bank in neuropixels

PSY-Yibo, 2023-06-19:


jsiegle, 2023-06-19:
Hi there! The electrode bank is stored in the `settings.xml` file which is saved along with each recording. Each probe has a section named `NP_PROBE`, which contains a `CHANNELS` entry that specifies the selected bank for each channel (with 0-based indexing on both banks and channel numbers).

You can also extract the electrode bank [from the exported IMRO files](https://billkarsh.github.io/SpikeGLX/help/imroTables/). 

The JSON files conform to the [ProbeInterface](https://probeinterface.readthedocs.io/en/main/) specification, and only includes site geometry (not bank info).

PSY-Yibo, 2023-06-20:
Thank you so much! I get it.

closed 2025-06-08 by jsiegle