open-ephys/plugin-GUI #577: New version of GUI not recognizing the old Acquisition board

HiroyukiKato7, 2023-07-07:
We have been using Open-Ephys 4.5 plugin GUI without any issues.  However, when we updated the GUI to the latest version, it failed to recognize our Open-Ephys Acquisition board.  We rolled back the GUI to the old 4.5 version, and it works without a problem.  
Is there a compatibility issue between the new GUI versions and the old acquisition boards?  My board was purchased back in September 2018.

jsiegle, 2023-07-07:
Just to check – are you using the Acquisition Board plugin from the "Rhythm Plugins," and not the "OE FPGA Acquisition Board" plugin?

HiroyukiKato7, 2023-07-07:
Yes, we have been using Rhythm FPGA Plugin.  Do we need to switch for the new GUI?

jsiegle, 2023-07-07:
In the latest version, the required plugin can be accessed by first installing the "Rhythm Plugins" via the Plugin Installer (ctrl-P), or selecting the default configuration (ctrl-D) for the Acq. Board. The plugin is now called "Acquisition Board" rather than "Rhythm FPGA." If you've already done those things, let us know.

HiroyukiKato7, 2023-07-07:
Thanks!  Now it recognizes the board.  However, I have a problem setting up the new GUI.
I have 2*64 channel probes connected to the board.  When I chose to save 128 channels plus two ADC, as you see in the attached screenshot, somehow the GUI saves ADC1-3 and ch2-128.  So, it saves ADC3 instead of ch1.  Do you have any idea why this happens?  It may be related to issue #438, but I am not sure.

[image]

jsiegle, 2023-07-07:
Can you double-check the ordering of the channel map? For example, it looks like channel 129 (ADC1) is mapped to channel 96. I think you only want to be reordering channels 1-128, and leaving the ADCs (129-136) in the original order.

HiroyukiKato7, 2023-07-07:
Oh, you are right.  It's weird, but it seems that all the values in the channel map shifts by 1 when I load the map that we used to use for GUI ver 4.5.  Has there been a change regarding something like 0-indexing?

jsiegle, 2023-07-07:
The channel map JSON files have been simplified, but I believe the indexing conventions should be the same as before. The latest format is documented here: https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Channel-Map.html#saving-and-loading-channel-mappings

HiroyukiKato7, 2023-07-07:
It seems that our older map starts at 1, instead of 0 in the current version.
Anyway, this solved my problems.  Thanks for your prompt answers!

closed 2025-06-08 by jsiegle