open-ephys/plugin-GUI #583: OSC plug in crashes GUI

cspurandare, 2023-08-08:
Hi all, we obtained our first neuropixel recordings (using open ephys gui) last week, where the stimulus was created and presented with bonsai. To sync these two, I tried using the OSC messaging, or ttls from bonsai routed to the 385th channel or both simultaneously. Although it did work sometimes, I noticed -

If both the TTL and osc were communicating, the open ephys gui would sometimes hang/crash when bonsai started. This crash was not accompanied by any error message in the log as far as I can tell. The crash was actually more of a "hang", where LFP and AP band signal just flattened out in the display and nothing was recorded anymore, but the gui showed that it's still recording (red bar at the top, timer still increasing). The CPU requirement tab was below 25-30% so I don't think it was being overwhelmed. I am recording on a 20 thread 10 core computer, using an SSD card, but looking at the SSD card performance, it sometimes does peak close to 100%.

On other occasions, the system worked, and I could see ttls from bonsai appearing in the last channel (385) in open ephys gui. Checking later the osc messages were stored ok too. But after running for 2 to 3 hours, the open ephys gui would randomly crash (flatlined, but gui shows the recording is still going on, as above).

I tried with 0.6.1 and 0.6.4 and saw these issues in both.
Has anyone else experienced this behavior and know of any solutions/fixes? Thanks in advance!

jsiegle, 2023-08-08:
Hi Chinmay – we are still looking into a fix for this; since the hanging happens randomly, it's been somewhat difficult to troubleshoot.

Given that the TTL outputs are being recorded by the Neuropixels event channel, do you actually need to send the OSC messages to Open Ephys? The OSC messages can be saved on the Bonsai side, and aligned to their respective TTL events offline.

The OSC Plugin is meant to facilitate closed-loop feedback based on analysis done in Bonsai – for example, gating the output of the Phase Detector when the subject enters a certain section of the environment. If this isn't what you're doing, the OSC Plugin is just sending redundant information.

jsiegle, 2023-08-08:
Also, if you can share the Bonsai workflow you've been using, it would be helpful for debugging.

cspurandare, 2023-08-08:
https://drive.google.com/file/d/1ewIk8jjGQ3wikIkw-W6MXnFwFff9PxJX/view?usp=drive_link, https://drive.google.com/file/d/1vUKrtjB7YLfVvkIFSM1lXZWYw7LcsGrV/view?usp=drive_link

I was not able to upload them here, so I am sharing the links above. Also, since the OP, I am now getting the issue every single time, in terms of the start of Bonsai crashes OpenEphys when the OSCEvents node is in the workflow. I have tried different ports (apart from 27020), different addresses (like/number, /ttl, /num etc), tried to create manual exceptions in the firewall for the ports, and 3 different versions of OpenEphys 6.0 6.1 6.5.
Thanks and do let me know if any other information might help get to the bottom of the issue.

jsiegle, 2023-08-10:
Hi Chinmay -- looking at the Bonsai workflow, I can see that the "OSCGenerator" group workflow is waiting 30 seconds, then sending a message every 200 ms containing the seconds value of the current software timestamp (which cycles through values between 0 and 59). On the Open Ephys side, this translates to 5 TTL "ON" events per second, with each second having events on a different TTL line.

This definitely falls outside of the intended usage of the OSC Events plugin. It's not meant to synchronize processes, because you can get much more precise synchronization by using hardware TTLs. Instead, it's designed to allow closed-loop feedback that combines analysis done in Bonsai (e.g. extracting ROIs from video) with analysis done in Open Ephys (e.g. detecting hippocampal ripples).

Given that you already have the ability to translate hardware TTLs into software times registered by Bonsai (using the "TTLGenerator" group workflow), the OSC Events is just adding redundant (and less precise) information. So I would recommend disabling this part of the workflow and removing the OSC Events plugin on the Open Ephys side.

We still want to determine what's causing these crashes, of course, so I am currently running this workflow + OSC Events plugin in debug mode to see what happens after a few hours. I will let you know what I find.

jsiegle, 2023-08-15:
The workflow + OSC Plugin are still running after ~5 days. Can you send the Open Ephys `settings.xml` file for your experiment, so we can try to replicate your exact configuration?

cspurandare, 2023-08-23:
<?xml version="1.0" encoding="UTF-8"?>

<SETTINGS>
  <INFO>
    <VERSION>0.6.0</VERSION>
    <PLUGIN_API_VERSION>8</PLUGIN_API_VERSION>
    <DATE>6 Aug 2023 19:54:38</DATE>
    <OS>Windows 10</OS>
    <MACHINE name="DESKTOP-NBQM1P2" cpu_model="Intel(R) Core(TM) i9-10900K CPU @ 3.70GHz"
             cpu_num_cores="20"/>
  </INFO>
  <SIGNALCHAIN>
    <PROCESSOR name="Neuropix-PXI" insertionPoint="0" pluginName="Neuropix-PXI"
               type="4" index="1" libraryName="Neuropix-PXI" libraryVersion="0.5.0"
               processorType="2" nodeId="100">
      <GLOBAL_PARAMETERS/>
      <STREAM name="19192703692-AP" description="description" sample_rate="30000.0"
              channel_count="384" device_name="Neuropixels 1.0">
        <PARAMETERS/>
      </STREAM>
      <STREAM name="19192703692-LFP" description="description" sample_rate="2500.0"
              channel_count="384" device_name="Neuropixels 1.0">
        <PARAMETERS/>
      </STREAM>
      <STREAM name="19192703742-AP" description="description" sample_rate="30000.0"
              channel_count="384" device_name="Neuropixels 1.0">
        <PARAMETERS/>
      </STREAM>
      <STREAM name="19192703742-LFP" description="description" sample_rate="2500.0"
              channel_count="384" device_name="Neuropixels 1.0">
        <PARAMETERS/>
      </STREAM>
      <CUSTOM_PARAMETERS/>
      <EDITOR isCollapsed="0" isDrawerOpen="0" displayName="Neuropix-PXI" activeStream="0"
              Type="Visualizer">
        <TAB Active="1" Index="3"/>
        <WINDOW Active="0"/>
        <NEUROPIXELS_EDITOR MainSyncSlot="0" SendSyncAsContinuous="0" SyncDirection="0" SyncFreq="0">
          <BASESTATION Directory="" Slot="4" NamingScheme="3" port1dock1="slot4-port1-1"
                       port1dock2="slot4-port1-2" port2dock1="slot4-port2-1" port2dock2="slot4-port2-2"
                       port3dock1="slot4-port3-1" port3dock2="slot4-port3-2" port4dock1="slot4-port4-1"
                       port4dock2="slot4-port4-2"/>
          <CUSTOM_PROBE_NAMES/>
        </NEUROPIXELS_EDITOR>
        <NP_PROBE slot="4" bs_firmware_version="2.0169" bs_hardware_version="UNKNOWN"
                  bs_serial_number="0" bs_part_number="NeurPix2 PXI Carrier" bsc_firmware_version="3.2176"
                  bsc_hardware_version="2.1" bsc_serial_number="21230805" bsc_part_number="NP2_QBSC_00"
                  headstage_serial_number="23040021" headstage_part_number="NP2_HS_30"
                  flex_version="4.13" flex_part_number="NP2_HS_30" port="3" dock="1"
                  probe_serial_number="19192703692" probe_part_number="PRB_1_4_0480_1"
                  probe_name="Neuropixels 1.0" num_adcs="32" custom_probe_name="19192703692"
                  ZoomHeight="100" ZoomOffset="0" apGainValue="500x" apGainIndex="3"
                  lfpGainValue="250x" lfpGainIndex="2" referenceChannel="Ext" referenceChannelIndex="0"
                  filterCut="ON" filterCutIndex="1" visualizationMode="0" activityToView="0">
          <CHANNELS CH0="0" CH1="0" CH2="0" CH3="0" CH4="0" CH5="0" CH6="0" CH7="0"
                    CH8="0" CH9="0" CH10="0" CH11="0" CH12="0" CH13="0" CH14="0"
                    CH15="0" CH16="0" CH17="0" CH18="0" CH19="0" CH20="0" CH21="0"
                    CH22="0" CH23="0" CH24="0" CH25="0" CH26="0" CH27="0" CH28="0"
                    CH29="0" CH30="0" CH31="0" CH32="0" CH33="0" CH34="0" CH35="0"
                    CH36="0" CH37="0" CH38="0" CH39="0" CH40="0" CH41="0" CH42="0"
                    CH43="0" CH44="0" CH45="0" CH46="0" CH47="0" CH48="0" CH49="0"
                    CH50="0" CH51="0" CH52="0" CH53="0" CH54="0" CH55="0" CH56="0"
                    CH57="0" CH58="0" CH59="0" CH60="0" CH61="0" CH62="0" CH63="0"
                    CH64="0" CH65="0" CH66="0" CH67="0" CH68="0" CH69="0" CH70="0"
                    CH71="0" CH72="0" CH73="0" CH74="0" CH75="0" CH76="0" CH77="0"
                    CH78="0" CH79="0" CH80="0" CH81="0" CH82="0" CH83="0" CH84="0"
                    CH85="0" CH86="0" CH87="0" CH88="0" CH89="0" CH90="0" CH91="0"
                    CH92="0" CH93="0" CH94="0" CH95="0" CH96="0" CH97="0" CH98="0"
                    CH99="0" CH100="0" CH101="0" CH102="0" CH103="0" CH104="0" CH105="0"
                    CH106="0" CH107="0" CH108="0" CH109="0" CH110="0" CH111="0" CH112="0"
[... 495 lines cut ...]
    <RECORD_NODE ID="103"/>
  </EDITORVIEWPORT>
  <DATAVIEWPORT selectedTab="4"/>
  <AUDIO sampleRate="48000.0" bufferSize="1024" deviceType="Windows Audio"/>
  <CONTROLPANEL isOpen="0" recordPath="C:\Users\Chinmay_Neuropixels\Documents\Open Ephys"
                recordEngine="BINARY"/>
  <AUDIOEDITOR isMuted="0" volume="50.0" noiseGate="0.0"/>
  <FILENAMECONFIG>
    <PREPEND state="0" value=""/>
    <MAIN state="1" value="2023-08-06_19-54-38"/>
    <APPEND state="0" value=""/>
  </FILENAMECONFIG>
  <PROCESSORLIST>
    <COLOR ID="801" R="59" G="59" B="59"/>
    <COLOR ID="804" R="241" G="90" B="41"/>
    <COLOR ID="802" R="0" G="174" B="239"/>
    <COLOR ID="803" R="0" G="166" B="81"/>
    <COLOR ID="805" R="147" G="149" B="152"/>
    <COLOR ID="806" R="255" G="0" B="0"/>
    <COLOR ID="807" R="0" G="0" B="0"/>
  </PROCESSORLIST>
  <UICOMPONENT isProcessorListOpen="1" isEditorViewportOpen="1"/>
</SETTINGS>

Hi sorry about the delay in response. Find inserted above

cspurandare, 2023-08-23:
[settingsFile_Copied.txt](https://github.com/open-ephys/plugin-GUI/files/12422649/settingsFile_Copied.txt)

jsiegle, 2023-08-25:
Thanks for sending! I ran this config for 6+ hours without observing any crashes. But we will keep trying to see if we can replicate the problem.

Some advice for now:
- Just perform synchronization with the hardware TTL line, which will be more precise than using OSC messages

For future use of the OSC plug-in for synchronization (in cases where hardware synchronization is not possible):
- Dedicate one TTL line for software synchronization, so this doesn't interfere with events on other lines
- Only send messages at longer intervals (e.g., 30-60 seconds). The potential clock drift on shorter timescales will be negligible.
- Tag each message with a software timestamp on the Bonsai side, and save these to a CSV

I'm going to close this issue, since the problem is also documented here: https://github.com/open-ephys-plugins/osc-io/issues/1

closed 2023-08-25 by jsiegle