open-ephys/plugin-GUI #584: Audio device initialization error on startup

cxrodgers, 2023-08-08:
Hello,
I have an Open Ephys Acquisition Board with the new FPGA. Trying to run on Linux .. not sure this is even supported with the new FGPA, but also I'm running into an issue even before I even get to that part. I installed the *.deb file for the latest version. When I start open-ephys, I get a graphical dialog window that says this:

```
Audio device initialization error
There was a problem initializing the audio device:
no channels
Retry / Quit
```
"Retry" repeats the message forever, and "Quit" quits. 

In the terminal, there is:
```
[open-ephys] Session Start Time: 8 Aug 2023 17:35:55
[open-ephys] Open Ephys GUI v0.6.5 (Plugin API v8)
[open-ephys] JUCE v6.0.8
[open-ephys] Operating System: Linux
[open-ephys] CPU: 12th Gen Intel(R) Core(TM) i5-12400 (12 core)
Killed
```
with "Killed" only appearing after I click "Quit" in the dialog.

I think this is likely related to (of course) audio issues in my Linux distribution -- I don't appear to have any audio output devices properly installed. However, I don't think this should be necessary to start Open Ephys, right?

thanks!

cxrodgers, 2023-08-08:
When I build from the current github source, I get a slightly longer error message in the terminal:
```
JUCE v6.0.8

[open-ephys] Session Start Time: 8 Aug 2023 17:41:25
[open-ephys] Open Ephys GUI v0.6.5 (Plugin API v8)
[open-ephys] JUCE v6.0.8
[open-ephys] Operating System: Linux
[open-ephys] CPU: 12th Gen Intel(R) Core(TM) i5-12400 (12 core)
[open-ephys][debug] Creating processor graph...
[open-ephys][debug] Creating audio component...
[open-ephys][debug] 
./Debug/open-ephys(_ZN4juce11SystemStats17getStackBacktraceEv+0x57) [0x55d8a4980701]
./Debug/open-ephys(+0xacf8ef) [0x55d8a4d9f8ef]
./Debug/open-ephys(+0xb7152f) [0x55d8a4e4152f]
./Debug/open-ephys(+0x6b07ff) [0x55d8a49807ff]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fd4bc8a1520]
./Debug/open-ephys(_ZN4juce6StringC2ERKS0_+0x1c) [0x55d8a4982342]
./Debug/open-ephys(+0xb72811) [0x55d8a4e42811]
./Debug/open-ephys(+0xafcaa8) [0x55d8a4dccaa8]
./Debug/open-ephys(+0xace43a) [0x55d8a4d9e43a]
./Debug/open-ephys(+0xb71c35) [0x55d8a4e41c35]
./Debug/open-ephys(+0xb71471) [0x55d8a4e41471]
./Debug/open-ephys(_ZN4juce19JUCEApplicationBase13initialiseAppEv+0xcc) [0x55d8a4a4322e]
./Debug/open-ephys(_ZN4juce15JUCEApplication13initialiseAppEv+0x1c) [0x55d8a4bf4a34]
./Debug/open-ephys(_ZN4juce19JUCEApplicationBase4mainEv+0xe0) [0x55d8a4a430b6]
./Debug/open-ephys(_ZN4juce19JUCEApplicationBase4mainEiPPKc+0xbe) [0x55d8a4a42f94]
./Debug/open-ephys(+0xb7018e) [0x55d8a4e4018e]
/lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fd4bc888d90]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fd4bc888e40]
./Debug/open-ephys(_start+0x25) [0x55d8a4818745]

Killed
```

anjaldoshi, 2023-08-08:
Hi Chris, 

The GUI needs to be able access an audio output device, otherwise it won't be able to function as all of the data acquisition callbacks are driven by the audio card.
If your system has an audio card installed, you either need to connect an audio output device, or you need to make sure the audio drivers are working properly. If you don't have an audio card, you'll need to install one.

cxrodgers, 2023-08-09:
got it, thanks! After I got audio output working, open-ephys will now start

closed 2023-08-09 by cxrodgers