open-ephys/plugin-GUI #586: Sawtooth timestamps.npy and disagreement with NIDAQ timestamps.npy for synced data streams

chris-angeloni, 2023-09-06:
I'm troubleshooting a recording setup with neuropixels spike recordings and video data. I'm using automatic synchronization within the recording node to generate synchronized timestamps, but I'm finding some strange behavior within the timestamp files that I am looking at for one recording.

Below I've plotted the timestamps (from timestamps.npy) for the AP recording node and the NIDAQ recording node.

[image]

[image]

I'm seeing a few troubling things:
1) The sawtooth shape seems like some clock is reaching its max value and resetting
2) There is a large discrepancy between the AP and NIDAQ timestamp values when the clock resets, and it seems to not resolve right away

One, is this expected behavior (I'd guess not, another recording I tried did not have this issue), two, how should I go about synchronizing this recording, if possible (I'm guessing I will need to revert to sample_numbers.npy, offset by the sample starts in sync_messages.txt?)

jsiegle, 2023-09-06:
What is the digital signal you're using for synchronization? If the Record Node synchronizer receives multiple events in the same buffer it will reset itself, leading to the behavior you're seeing. Can you send the `events` directory for this recording? That should make it clear what's going on.

To get the true timestamps out for this session, you'll need to perform synchronization offline using `sample_numbers.npy` files.

chris-angeloni, 2023-09-06:
It is just a 1Hz TTL pulse from an arduino sent to the NPX sync channel, duplicated to a NIDAQ digital channel, and also to some LEDs. This exact setup worked for a different recording in the same session.

[events.zip](https://github.com/open-ephys/plugin-GUI/files/12542718/events.zip)

jsiegle, 2023-09-06:
Thanks for sending! I'm not seeing anything obvious that could have caused the periodic resetting. Although I did notice that the very first event timestamps do not match across streams, so something was off with synchronization before the recording even started. If the sync monitors appeared green, that's a bug we'll need to track down.

The good news is that the `sample_numbers.npy` files are totally fine, and can be used to perform synchronization offline.

chris-angeloni, 2023-09-06:
Ok, great. Per issue #502, I have a healthy distrust of the sample start times in sync_messages.txt. Can you confirm whether this issue has been fixed?

Thank you for your help!

jsiegle, 2023-09-06:
All issues with `sync_messages.txt` were fixed as of GUI version 0.6.0...we completely rewrote the code that was generating this file.

paolahydra, 2024-05-02:
@jsiegle 
I apologize beforehand; I am going to ask a vague/imprecise question because I don't know where I read this. I seem to remember that the NPX2 data is not sampled exactly (or uniformly?) at 30KHz, and therefore, we should rely more on the saved timestamps... 
Would this be a problem then?
Thanks!

paolahydra, 2024-05-02:
If it helps, this is what the timestamps look like in one of our recordings:
[image]

jsiegle, 2024-05-02:
Unfortunately the online synchronizer does not work well with barcode sync pulses. It looks for coincident events across streams, and if there are many events in a short interval this can make it impossible to align the pulses precisely, leading to strange timestamp sequences like what you observed. The solution is to ignore the online timestamps and recompute the global timestamps offline, using the values in `sample_numbers.npy`, rather than `timestamps.npy`.

Within the next few months, we plan to make some changes to the online synchronizer so it works with any type of input. This will involve allowing it to recognize patterns of event transitions, rather than trying to align to single pulses only. We also plan to add more visual feedback about the state of the synchronizer, so it's easier to catch cases where the timestamps are not increasing linearly.