open-ephys/plugin-GUI #587: crash by buffer is full

FabianMunozSilva, 2023-09-13:
Hello,
We are using the 0.6.5 version of the open ephys gui. The gui stopped recording because the buffer was full. We are using neuronexus 32channels and we acquired them as binary, but as another record plugin, we acquired the analog signal (8ch) in openephys format for the reconstruction of the trials. After 25 min stopped recording because the buffer was full in the 8 analog signals recorded. The problem just started before we didn't have this issue. We restarted the PC, and reconnected the cables. and then reconnected the box, but the problem still is there.
I hope you can give us some ideas to solve it
Thanks in advance

medengineer, 2023-09-13:
Hello,

To clarify, this issue arose suddenly without any changes to the GUI/plugin version, hardware, or signal chain?

Can you send the signal chain .xml file to [pavel@open-ephys.org](mailto:pavel@open-ephys.org)? 

Thanks!

FabianMunozSilva, 2023-09-14:
Hi, 
Yes, that happens without any change in the openephys-gui, we have been using version 0.6.5 for several months. 
Thanks, I will send the email to pavel