open-ephys/plugin-GUI #588: Error running Open Ephys GUI after compiling

ViniciusSilva96, 2023-09-19:
I am new to C++ coding and applications for Windows so I don't know what I am doing wrong here.

I had to change the order of the filter in the filter plugin so that it could be used for our use case.  To do that I downloaded the repository, changed it in Visual Studio, and compiled it following the instructions in the guides, so far so good. But as I am trying to run the application I get the error: 

"Point of entry not found. ... ??0this_will_fail_to_link_if_some_of_your_compile_units_are_built_in_release_mode@juce@@QEAA@XZ... "

So I tried to change the propriety values to 0 in the CMakeList.txt file:
`set_property(DIRECTORY APPEND PROPERTY COMPILE_DEFINITIONS
	$<$<PLATFORM_ID:Windows>:_CRT_SECURE_NO_WARNINGS>
	$<$<PLATFORM_ID:Windows>:NOMINMAX>
	$<$<CONFIG:Debug>:DEBUG=1>
	$<$<CONFIG:Debug>:_DEBUG=1>
	$<$<CONFIG:Release>:NDEBUG=1>
	JUCE_APP_VERSION=${GUI_VERSION}
	JUCE_APP_VERSION_HEX=${GUI_VERSION_HEX}
	)`

But it is still not working. I also tried compiling the unmodified version of the GUI and got the same error.

anjaldoshi, 2023-09-19:
Hi - Can you confirm you followed all the instructions mentioned in the docs here: https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html#windows, including the instructions in the "Tip".

Also, if you're planning to use the custom version for experiments, you should build the GUI in "Release" mode if not already. You can change the Build type from the toolbar at the top in Visual Studio. 

[image]

If that still doesn't help, can you try running the GUI executable (open-ephys.exe) directly. The executable is located under `plugin-GUI\Build\Debug` or `plugin-GUI\Build\Release` depending on the build type.

ViniciusSilva96, 2023-09-19:
It worked, was the "Release" mode in the Visual Studio. 

Thank you for your time.

closed 2023-09-20 by ViniciusSilva96