open-ephys/plugin-GUI #591: Disabling channel 1 in channel mapper zeros all channels

aacuevas, 2023-11-20:
In the channel mapper, disabling the channel marked as "1" causes all the channels to be of zero value. This is independent of the position on the "1" channel in the map.

jsiegle, 2023-11-20:
This is due to the way that we are connecting channels across processors. JUCE's built-in method for wiring up the processor graph check for connections between every possible combination of channels across all processors, and is therefore extremely slow for high channel counts. To speed this up, we modified it to only check the first channel. However, this means the Channel Map won't get connected if channel 1 is disabled (even if you change the channel order).

The wiring algorithm is fortunately much more efficient in JUCE 7, so this will get fixed in the next major GUI release.

jsiegle, 2025-06-08:
Confirming this is fixed in v1.0.

closed 2025-06-08 by jsiegle