open-ephys/plugin-GUI #592: “Apply to all” appears to, but does not actually switch channels being recorded in the GUI and the settings.xml.

egmcbride, 2023-12-04:
Using OpenEphys 0.6.6 and Neuropixels-PXI, we set the middle bank of channels (384-768) to be recorded by loading a saved .imro file for a single probe, and verified the signals on that probe are noticeably different in the way we expect, i.e. a clear in- to out-of-brain transition for a deep insertion. 

Then after pressing “apply to all”, the new channel banks appear in the GUI to be selected on the other probes, but on closer inspection, the signals have not changed, and they are still recording from the original channels (the lowest 384). Then, when loading the .imro file for each individual probe, the signals do change in the way expected. When recording after pressing apply to all, the settings.xml file reflects what is shown in the GUI, the incorrect channels.

It would be very easy to miss this if you aren't looking closely at the signals or inspecting the raw data.

jsiegle, 2023-12-06:
Thanks for posting this! It's been tricky to debug because we haven't been able to replicate this behavior on any of our dev machines. Which rig were you seeing this on? Can we run some tests on it when it's free?

bjhardcastle, 2023-12-06:
A while back we found that remote configuration of the channels on this rig (NP.3) also wasn't working, where it works fine on other rigs with the same version and same HTTP request.

One obvious difference on the rig is a NI-DAQmx source in the signal chain, and a Merger with the Neuropix-PXI source - I wonder whether that could be the cause?

egmcbride, 2023-12-06:
Also, NP3 will be free this Friday afternoon (12/8) if you want to run some tests then.

jsiegle, 2023-12-21:
In the latest version of the Neuropixels plugin (available now), pressing the "Apply to all" button displays a progress bar until all of probe settings have been updated. It also avoids re-applying the settings for the currently active probe, which should make the update happen a little bit faster.

In addition, you can now select preset electrode configurations from a drop-down menu, so it's much easier to switch between different channel banks.

bjhardcastle, 2023-12-21:
Thanks a lot Josh, the dropdown is great!

bjhardcastle, 2023-12-21:
We'll update on whether the channels actually switch for us using this method.

Apparently waiting longer didn't work.