open-ephys/plugin-GUI #593: GUI crashing

ShivaniBITS25, 2023-12-17:
Hey, I am a newbie to Open Ephys.  GUI not responding and shows a dialogue box to mail to the support group. I have attached the screenshots. Sometimes it started to acquire the data, and it stopped after 1-2 seconds. Please let me know how we can resolve this issue.

Thanks,
Shivani

[image]

jsiegle, 2023-12-17:
Hi @ShivaniBITS25! A number of users with Open Ephys acquisition boards have experienced similar issues in the last few days, due to an automated Windows driver update that broke compatibility with our hardware.

If you can email `support@open-ephys.org`, we can send you details about how to resolve it. 

We will post information about a general solution once we have a better idea about exactly what's causing this problem.

ShivaniBITS25, 2023-12-18:
Hey Josh, Thanks for the quick response. I already sent the mail regarding this issue to your team member. I will wait for the response.

ChucklesOnGitHub, 2023-12-18:
If you have an OE FPGA Acquisition Board and are using the Open Ephys GUI on Windows, you might have recently been experiencing frequent crashes that impede data acquisition.

A recent driver update to version 1.3.0.8 is not compatible with our hardware. The driver is being automatically updated by Windows. We are still investigating the cause of the incompatibility but the current fix is to install version 1.3.0.4 of the driver.

Important: This fix is for the error reported in the GUI console window as  `***ERROR*** Error reading ONI frame: Uknown error code -largenumber` which causes acquisition to stop and the GUI to stop working and shut down.

[image]

If that is the case for you, please follow these instructions:

1. With the acquisition board connected, check the driver version in the Device Manager: find the acquisition board under Universal Serial Bus controllers as FTDI FT600 USB 3.0 Bridge Device and double-click it to access its properties. 
[image]

2. Go to the “Driver” tab and note the Driver Version you have.

[image]

3. If your driver version is 1.3.0.8, click on Uninstall Device and check the box in the pop-up window to remove the driver software.
4. Install version [1.3.0.4](https://www.ftdichip.com/Driver/D3XX/FTD3XXDriver_WHQLCertified_v1.3.0.4_Installer.exe.zip) of the driver (this is a direct link that should start the download for you). Be wary of future automatic Windows updates for this driver.
5. Check that acquisition in the GUI works.

If the error code you get is code: -5 or another small number, or you continue having issues after checking that the driver version is 1.3.0.4, and version 1.3.0.8 is uninstalled, please let us know: send your operating system details, driver version and the complete contents of the GUI console window to `support@open-ephys.org` so we can continue troubleshooting.

ChucklesOnGitHub, 2023-12-18:
Please follow [this issue](https://github.com/open-ephys/ECP5U85-BSE-USB/issues/3) for developments.

closed 2024-01-29 by ChucklesOnGitHub