open-ephys/plugin-GUI #596: noise can not be removed

zkcsmxz, 2024-01-08:
[image]

Hi
 when i use OE to record, i suddently has strong 50 hz noise which can not be remove. 
I tried different OE board, differen headstage and animal. it turn out same result. 
I ground the Faraday Cage, holder and everything.

can any one help me 
thanks a lot!

jsiegle, 2024-01-09:
A couple things to try:
- Make sure the acquisition board is grounded (you can clip to the BNC cable on the back)
- Completely unplug any devices near the animal (sometimes devices can cause noise even when powered off)
- Add a [Common Avg Ref](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Common-Average-Reference.html) plugin after the Record Node but before the LFP Viewer. This is not as good as removing the physical sources of the noise, but it may be sufficient for getting rid of the 50 Hz artifact.

ipataanna, 2024-06-11:
Hi jsiegle,
I just moved to Open Ephys for single neuron recording in non human primates and I am learning... 
can you please explain me how does the Common Avg Ref do?

thank you
Anna

jsiegle, 2024-06-16:
Hi Anna – I just added some more information to the Common Average Ref docs page: https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Common-Average-Reference.html

Can you read that through and let me know if you have any additional questions?

closed 2025-06-08 by jsiegle