open-ephys/plugin-GUI #597: Bug in channel group sorting for most NPX2 multishank configurations.

paolahydra, 2024-01-25:
Hi Josh,

We are working with the latest versions of the API (v0.6.6) and Neuropix PXI plugin (release 0.6.2). Running on Windows 10 Enterprise, version 2004.

We have noticed significant inconsistencies in the Probe Viewer and LFP Viewer channel ordering when selecting most multishank NPX 2.0 channel configurations, where the signal presented does not match the corresponding channel label (and depth).

In Probe Viewer, only when we split the channels across the four shanks (i.e., All Shanks 1-96 configuration) do we get a reasonably correct-looking signal display like this: 
[image]

For all the other configurations we have tried, the channel labels are always correctly sorted, but the corresponding signals do not match the labels, resulting in scrambled chunks of 48 channels (see example images below from 2- and 1-shank configurations, where spikes were only present in ~ the lower half of each shank). 
Within each block of 48, signal sorting is fine.
In the saved data, channel/signal sorting is also fine.

Also, the LFP Viewer and Probe Viewer signals do not match (e.g., a channel labeled with the same name may have spikes in one viewer but not the other). Note that we are not confident about the correct sorting of the LFP Viewer channels either (but haven't checked carefully enough).

Thank you for looking into this.

[image]
[image]

jsiegle, 2024-01-26:
Hi @paolahydra, thanks for documenting this in detail. It turns out there was a bug in how the Probe Viewer channels were being sorted for visualization. The channel numbers along the left side were correct, but these were not mapped to the correct continuous signals. That explains the discrepancy between the signals in the Probe Viewer and LFP Viewer.

We will release a new version of the Probe Viewer today that implements this fix.

paolahydra, 2024-01-27:
Thank you for taking care of this!