open-ephys/plugin-GUI #598: Distributing FTDI driver v1.3.0.8 with the GUI

ChucklesOnGitHub, 2024-01-29:
[Installers included in the GUI](https://github.com/open-ephys/plugin-GUI/blob/6111d120968eb5f336769b4e8366f17637bc682e/Resources/Installers/Windows/windows_installer_script.iss#L29) point to FTDI driver version 1.3.0.4, but they are now on version 1.3.0.8. Problems originally caused by driver version 1.3.0.8 are [resolved](https://github.com/open-ephys/ECP5U85-BSE-USB/issues/3) in OE FPGA Acquistion Board plugin v 0.2.3.

Is this stable enough to package driver version 1.3.0.8 with the GUI?

jsiegle, 2024-01-29:
@aacuevas what are your thoughts on this?

aacuevas, 2024-01-30:
It should be ok to go ahead and distribute the newer driver

anjaldoshi, 2024-02-01:
Done in https://github.com/open-ephys/plugin-GUI/commit/4b8408e46752425c8c2a245acad9e3482f8cfae6.
New downloads of GUI v0.6.6+ for Windows are now packaged with driver v1.3.0.8

closed 2024-02-01 by anjaldoshi