open-ephys/plugin-GUI #599: Issues with opening the plugin installer

nathanielnyema, 2024-01-29:
I've been running into issues recently with opening the plugin installer from the GUI. I keep getting the following error in the console:

```
[open-ephys] Beginning HTTP server on port 37497
[open-ephys] Checking for a newer version of the GUI...
[open-ephys] ***ERROR*** Unable to fetch plugins! Please check your internet connection and try again.
[open-ephys] ***ERROR*** Unable to fetch updates! Please check you internet connection and try again.
```

This computer is running Windows 11. The internet connectivity is fully intact (I'm using it to submit this issue at the moment). I also tried this on a different computer running MacOS Ventura and connected to a different Wifi network and got the same error. Is it possible that there's an error on the server side with the plugin installer?

anjaldoshi, 2024-01-29:
Hi - I just verified that both of  the services necessary for the Plugin Installer to work are up and running. It could be because your Network Admin has blocked access to certain  endpoints automatically and that's why it is not working for you. Can you try to access these two URLs directly via your browser using the same machine and see if they are loading without any errors for you?
https://openephys.jfrog.io
https://open-ephys-plugin-gateway.herokuapp.com

nathanielnyema, 2024-01-29:
I'm able to access the first link but not the second. I was able to trace it back to an issue with our network and I'm reaching out to our network admin about it. Thank you!

closed 2024-01-29 by nathanielnyema