open-ephys/plugin-GUI #600: Opal Kelly Acquisition board not found

AnnaB27, 2024-02-08:
Hi,
I am using OphenEphys, version 0.4.6 with an Opal kelly XEM6310-LX45 Acquisition Board and I have Windows 11 pro on my computer.
I already recorded some data without any problems a few months ago. After december we stopped the recordings, but recently I don't know why but, when I tried to open OE, it gives me the error "Acquisition board not foud. Connect one". Since everithing is still the same as before and we didn't touch anything during the break, I don't understand what's happening.
I guess it could be a cable problem? In that case I am already looking for oneother cable to try and see what happens.
Could it be a drivers issue? maybe following some windows update o something similar?
Or could it also be directly a sign the the Acquasition boards has hardware problems and maybe I need to buy a new one? 

Thanks in advance for the help
Anna

anjaldoshi, 2024-02-09:
Hi Anna,

I'm not sure what could be the exact reason. As v0.4.6 is pretty old and we have made significant updates to the GUI as well as the Acquisition Board plugin since then, I'd recommend you to try the latest version of the GUI (v0.6.6) and then install the Acquisition Board plugin by installing the Rhythm Plugins using the Plugin Installer (refer the relevant docs [here](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Acquisition-Board.html)). See if that works.

Testing with multiple USB cables and/or computers would also be helpful.

-Anjal

AnnaB27, 2024-02-10:
Hi Anjal,
Thank you for the help,
I know that we are using a pretty old version, but when we tried working with the last version 0.6.6 we had some problems: we could not record our data (256 channels) in openephys format with multiple recording node. It was long ago so I can recall the exact problem but I remember that the session would crush or the only format accepted was binary. And that's why we went back to a version we knew would work. 
Anyway, Thank you again