open-ephys/plugin-GUI #601: Record Node directory does not update unless '+' button is manually pressed to start a new data directory

lachioma, 2024-02-14:
Hello

I had an unpleasant behavior of Open Ephys and Record Node that I think should be handled better to prevent data loss.

This is what happened step by step.

1. Open Ephys GUI was running an acquisition and recording data (=writing data to disk). 
2. At some point, I got a warning that there was no space in the disk any more, and the data recording stopped while the acquisition remained on (all good so far).
3. I then changed the directory of the Record node to another empty disk, without interrupting the acquisition. I started recording data again (the control panel of the GUI turned red), but the data was actually writing on the old directory without me realizing.
4. The disk got full very soon (5-6 extra GBs of data) until no more data could be saved, but no warning appeared and the GUI had always remained red in record mode until the end of the experiment. No data was saved in the updated directory, so I lost the data from second part of the experiment :(

I then ran a test recording, and found that changing the Record Node directory after interrupting the acquisition resulted in the same behavior, i.e. data was saved in the old directory instead of updating to the new one.
I found out that after setting a new directory in the Record Node, I should have clicked on the '+' button of the GUI control panel to start a new data directory, and finally start recording.

Suggested behavior: When changing the Record node directory, the data directory in the control panel should update automatically. Or throw some warnings. In any case, the GUI should definitely not be in Record mode while data is not actually being written to disk.

jsiegle, 2024-02-14:
Currently, the data directory will not change until the create new directory ("+") button is pressed. But I agree that expected behavior should be to always update the directory if the Record Node's root directory is changed. I think this can be fixed by adding a call to `RecordNode::createNewDataDirectory` whenever [`RecordNode::setDataDirectory`](https://github.com/open-ephys/plugin-GUI/blob/4b8408e46752425c8c2a245acad9e3482f8cfae6/Source/Processors/RecordNode/RecordNode.cpp#L314-L320) is called, but we will test this to make sure.

And it is indeed unfortunate that `RecordNode::checkDiskSpace` checks the currently selected data directory, rather than the directory where the data is actually being written. We will fix this as well!

jsiegle, 2024-03-01:
These issues are fixed in the latest release.

closed 2024-03-01 by jsiegle