open-ephys/plugin-GUI #602: OE FPGA Acq Board plugin impedance measurement fails after loading a previous config but works from empty signal chain

ChucklesOnGitHub, 2024-02-22:
User using OE acquisition board with an Intan headstage can get impedance measurements but sometimes the impedance measurement fails.

When the impedance measurement fails, the loading bar of the Impedance measurement just jumps from 0 to 100 and no measurements are taken. See video.
https://github.com/open-ephys/plugin-GUI/assets/16124331/7cbf01e2-5e6f-4dc3-bced-bd9970cc549d

User says: This impedance measurement error seems to occur whenever Open Ephys loads the lastConfig.xml file or the recoveryConfig.xml file. I had noticed that I tend to close open ephys by closing the Open Ephys GUI window (instead of the console window). After reopening the GUI, it automatically loaded the lastConfig.xml file and the measurement didn’t work anymore. When closing the GUI by closing the console window and then reopening, this window appears:
[image]

Only when I clicked “empty signal chain” and build the signal chain again, did the Impedance measurement work. When I clicked “recovery config” or “last config”, it of course loaded the recoveryConfig.xml file or lastConfig.xml file and the impedance measurement did not work anymore.

See attached document for detailed troublehooting report from the user, console log marked in red whenever it informs that a config file was loaded and with a note at the end if the Impedance measurement worked or not. Not working means showing the weird behavior in the video.

[recreatingImpMeasError.docx](https://github.com/open-ephys/plugin-GUI/files/14374388/recreatingImpMeasError.docx)

jsiegle, 2024-02-23:
I figured out what's going on...the first time the plugin is added, the variable that holds the DSP cutoff frequency in the plugin editor is not initialized, and this gets saved as a very high or very low value. If you look at the OE FPGA Acquisition Board plugin in the video, you can see three dots next to the "DSP" button, indicating a value so high it can't be displayed. This prevents impedance measurement from running. Changing that to a lower value (e.g. 0.1 or 0.5) should make it work again, and will also ensure the value is saved properly in the config file.

I fixed the initialization issues in [this branch](https://github.com/open-ephys-plugins/rhythm-oni-plugin/commits/fix-dspcutoff-saving/). @anjaldoshi can you put out a new release of this plugin?

anjaldoshi, 2024-02-24:
:white_check_mark: Merged and released as v0.2.5

shababo, 2024-03-11:
Hi, I've discovered what might be a similar issue. It seems running "Measure Impedances" does not work when I set the sampling rate to 5.0 kS/s. I tested by running the measurements first at the default 30.0 kS/s which worked. Then switching to 5.0 kS/s resulted in the measurement window hanging at 0%. I could then cancel this, switch back to 30.0 kS/s, and it worked again.

Also, while I'm posting, is it possible to run the impedance measurements 1) for only a subset of the channels and 2) via the python API/REST API?

Thanks so much!

jsiegle, 2024-03-12:
Thanks for posting, we will look into why it's failing for lower sample rates.

Those are great feature requests, I just created issues for them [here](https://github.com/open-ephys-plugins/rhythm-oni-plugin/issues/18) and [here](https://github.com/open-ephys-plugins/rhythm-oni-plugin/issues/17).

shababo, 2024-03-12:
Awesome! I really appreciate it. I'm happy to help with some of these. Would love to be able to PR additions to the REST API to allow for full programmatic control, and generally add functionality as needed.

jsiegle, 2024-03-14:
That's great! To add the ability to remotely change parameters or trigger impedance measurement, you'd need to implement the `handleConfigMessage` function in the `DeviceThread` class. You can use [the same function in the Neuropixels plugin](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/4c50364997d94490b8652f552e8ea2690360708f/Source/NeuropixThread.cpp#L1236-L1392) as a guide. If you get stuck on anything, don't hesitate to open an issue in the `rhythm-oni-plugin` repo!