open-ephys/plugin-GUI #604: Open-Ephys added a strobe bit and now my LFP streamline in the record node is not synchronizing correctly.

mackatar22, 2024-03-15:
Hello,

This is Mack from Boston University working with the neuropixel and openephys software. My PI and I recently added a strobe bit to the ttl box that goes into our digital/analog breakout box which we are using digital for. We had been running the open ephys before without a strobe bit however we recently moved one of our event bits to the strobe bit and for some reason that has caused some weird synchronization with the LFP streamline in our two record nodes only on the LFP streams. If anyone has had this issue or knows where to look any advice would be greatly appreciated. Thank you. Also for some reason if I don't get an update (still need to verify my email for github) then please email me at mackatar22@gmail.com or mjf22@bu.edu to get back to me if this ticket doesn't work.

medengineer, 2024-03-15:
Hi Mack,

What exactly is the strobe bit sending? If the pulses are too short online synchronization will not work. 

Can you send your signal chain to [pavel@open-ephys.org](mailto:pavel@open-ephys.org) just in case? (via File -> Save As in the GUI). 

Thanks!

medengineer, 2025-06-09:
Follow up email revealed failing probe tests that likely caused CPU spiking and affected proper synchronization. Regardless, v1.0 synchronizer has been upgraded to support arbitrary synchronization signals.

closed 2025-06-09 by medengineer