open-ephys/plugin-GUI #606: Open ephys to intan RHA

anacastropsico, 2024-03-19:
Hello, I have itan's oldest registration system, RHA, I would like to know if it is possible to use open ephys with it, as I saw a possibility only for RHD.

jsiegle, 2024-03-19:
Is this the system you'd like to use?

[image]

This was supported in the earliest versions of Open Ephys software, but we no longer maintain a plugin for it.

anacastropsico, 2024-03-19:
Yes this is. But where I can to access this versions? Is similar to Iran demo?

jsiegle, 2024-03-19:
Here's the source code for that version: https://github.com/open-ephys/gui

You'll need to compile it yourself, and you'll definitely encounter some bugs. If you already have the Intan demo software working, I'd recommend using that.

anacastropsico, 2024-03-21:
You believe be possible use this version you send me?

jsiegle, 2024-03-21:
Is there anyone in your group who knows C++ and could try to compile the software? This version should work to acquire from the RHA device, but unfortunately we can't support it because it's 8 years old at this point.

anacastropsico, 2024-03-21:
yes I'm trying with friends who are c++ programmers.
Thank you for your response and attention