open-ephys/plugin-GUI #607: GUI crashes when "headphones" is set as sound output

ChucklesOnGitHub, 2024-03-22:
I've ran into this issue before, and everything works fine when I switch to speakers in the windows sound control bar.

Here is an activity log from a user that resolved it the same way.

I can provide more troubleshooting if you can't recreate the issue.

[activity_2024-03-20_15-29-25.log](https://github.com/open-ephys/plugin-GUI/files/14722742/activity_2024-03-20_15-29-25.log)

ChucklesOnGitHub, 2024-04-23:
Just recreated this issue. The GUI closed when I dragged the File Reader into the Signal Chain while the sound source was set to bluetooth headphones. This did not happen with wired headphones.

[activity_bluetooth headphones_file reader.log](https://github.com/open-ephys/plugin-GUI/files/15075863/activity_bluetooth.headphones_file.reader.log)
[activity_wired headphones_file reader.log](https://github.com/open-ephys/plugin-GUI/files/15075864/activity_wired.headphones_file.reader.log)

magbaeta, 2024-04-26:
Just so you are aware, my bluetooth headphones did not cause this issue. Let me know if you need me to troubleshoot anything.

jsiegle, 2025-06-08:
We have added some extra checks in v1.0 that should prevent this from happening. If you encounter it again, please open a new issue.

closed 2025-06-08 by jsiegle