open-ephys/plugin-GUI #608: EventChannel Metadata is not saved

cmmoenne, 2024-04-10:
Hi,

I'm updating an old plugin because we revamped our hardware in the lab, so we need to use the new OE GUI. The old GUI allowed metadata to be saved from event channels, but it doesn't work in the current one. I looked at the code, and that functionality is now commented. Is it not working? Do you recommend any alternative?

Thanks for any help!

jsiegle, 2024-04-10:
There were very few plugins that used event metadata, and a lot of work was required to enable saving of arbitrary metadata types, so we deprecated this feature in more recent versions of the GUI. 

What is the specific metadata you'd like to save? Could it be encoded as [text events](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html#text-events) instead? Those are saved automatically for all data formats.

cmmoenne, 2024-04-11:
I'm going to use text messages then (using the getMessageChannel() channel).

**_What is the specific metadata you'd like to save?_**
We have to log 15 variables represented by integers and floats. I will make a CSV-style message to save them in a single message. Saving floats as text is not the best, but it will work perfectly for us.

Thanks!

closed 2024-04-11 by cmmoenne