open-ephys/plugin-GUI #609: Issue installing NI-DAQmx plugins

Olivier-Dub, 2024-04-24:
Hi! 
I tried to install NI-DAQmx plugins (Open Ephys 0.6.7) but I have the following error message:
"***ERROR*** Failed to load plugin DLL. Error code 126: The specified modules could not be found"
Any ideas?

Thanks!

medengineer, 2024-04-24:
Hi,

Are the NIDAQmx drivers installed? 

https://open-ephys.github.io/gui-docs/User-Manual/Plugins/NIDAQmx.html#requirements

Olivier-Dub, 2024-04-24:
Hi! Thanks for your answer. Yes the version 2024 Q2 is installed.

medengineer, 2024-04-25:
Hmm, do other installed plugins load fine?

Are you able to communicate with your NI device(s) via NI MAX or LabView?

Olivier-Dub, 2024-04-25:
I reinstalled Ni MAX and all NI dependencies, it seems to work now, NI-DAQmx plugin is install... I don't know what was wrong the 1st time. Thanks again for your quick answers though.

closed 2024-05-22 by medengineer