open-ephys/plugin-GUI #611: Cannot open signal chain with OE FPGA processor without acquisition board being plugged in

magbaeta, 2024-04-26:
I was working with the OE FPGA acquisition board using the OE FPGA acquisition board source processor without an issue.
Next time I tried to use the GUI without the board plugged in, the message “Open Ephys has stopped working” appeared.
[image]
[activity_2024-04-23_16-54-22.log](https://github.com/open-ephys/plugin-GUI/files/15133097/activity_2024-04-23_16-54-22.log)
It seems like the OE FPGA acquisition board needs to be plugged in for the GUI to run correctly.
This problem does not occur with the original source processor. With the original acquisition board it is possible to open a signal chain with the source processor without having to have the hardware plugged in. The message “Acquisition board not found” appears, but the GUI does not shut down, and you can press Cancel to continue inspecting the signal chain (even though you cannot start acquisition).
I’ve ran into the same issue on 2 different computers. Being able to open the signal chain regardless of the hardware that is plugged in is useful for troubleshooting.
[image]
[image]

jsiegle, 2024-04-26:
This is something we'll definitely want to fix. I opened a separate issue to track it in the `rhythm-oni-plugin` repo: https://github.com/open-ephys-plugins/rhythm-oni-plugin/issues/20

jsiegle, 2025-06-08:
This has been fixed in the OE FPGA Acquisition Board Plugin.

closed 2025-06-08 by jsiegle