open-ephys/plugin-GUI #612: Startup errors when launching GUI

SusanL82, 2024-05-20:
Hi everyone, 
We recently installed OpenEphys for our brand new neuropixels setup (following the installation guide on the openephys website). We are getting a number of errors on startup, even though everything seems to work just fine. 
Is this something we should worry about? And if so, what should we try to do about it? 
Best, 
Susan

```
[open-ephys] Session Start Time: 20 May 2024 16:30:03
[open-ephys] Open Ephys GUI v0.6.7 (Plugin API v8)
[open-ephys] JUCE v6.0.8
[open-ephys] Operating System: Windows 10
[open-ephys] CPU: 12th Gen Intel(R) Core(TM) i7-12700 (20 core)
[open-ephys] Audio device name: Reproduktory (Realtek(R) Audio)

[open-ephys] Audio device type: Windows Audio
[open-ephys] Audio output channels: 2
[open-ephys] Audio device sample rate: 44100
[open-ephys] Audio device buffer size: 1024

[open-ephys] ***ERROR*** Failed to load plugin DLL. Error code 193: %1 is not a valid Win32 application.

[open-ephys] ***ERROR*** WdfCoInstaller01009.dll Load FAILED
[open-ephys] ***ERROR*** Failed to load function 'getLibInfo'. Error code 127: The specified procedure could not be found.

[open-ephys] ***ERROR*** FpgaManager.dll Load FAILED
[open-ephys] ***ERROR*** Failed to load function 'getLibInfo'. Error code 127: The specified procedure could not be found.

[open-ephys] ***ERROR*** FTD3XX.dll Load FAILED
[open-ephys] ***ERROR*** Failed to load function 'getLibInfo'. Error code 127: The specified procedure could not be found.

[open-ephys] ***ERROR*** NeuropixAPI_x64_1_20.dll Load FAILED
[open-ephys] ***ERROR*** Failed to load function 'getLibInfo'. Error code 127: The specified procedure could not be found.

[open-ephys] ***ERROR*** NeuropixAPI_x64_3_62_1.dll Load FAILED
```
After this, we get messages about our started chassis, headstage, probe etc.

anjaldoshi, 2024-05-20:
Hi Susan,

Looks like the Neuropixels-PXI plugin installation was somehow messed up as the GUI is trying to load the dependencies of the plugin as separate plugins. There is also a `WdfCoInstaller01009.dll` file which shouldn't be there in the first place.

Can you share a screenshot of the following folders? That way we can make sure why are these files being loaded as plugins.

`C:\ProgramData\Open Ephys\plugins-api8`
`C:\ProgramData\Open Ephys\shared-api8`

ProgramData is a hidden system folder, so simple copy paste the above paths in the File Explorer address bar to access them.

SusanL82, 2024-05-21:
The plugin-api8 folder looks like this: 
[image]

and the shared-api8 folder like this: 
[image]

anjaldoshi, 2024-05-21:
Both the folders look good. Nothing out of the ordinary. The plugin and its dependencies are installed correctly here, which makes sense as you're able to load and use the Neuropixels plugin. 

I'd be surprised if the "plugins" folder next to the GUI executable contains the above-mentioned files. This should only contain plugins that are shipped with the GUI. Can you share a screenshot of it as well? If you installed the GUI via the installer and used the default location, it should be located here: `C:\Program Files\Open Ephys\plugins`

anjaldoshi, 2024-05-21:
Yeah, go ahead and delete the entire Enclustra_Windows_10 folder, FpgaManager.dll, FTD3XX.dll, and the NeuropixAPI_X64 dlls from that folder. After that, your errors should be gone. 

You don't have to add/copy files manually to that  folder when you want to install any plugins. You can use the Plugin Installer (Ctrl +P) to install/update/remove any plugin.

SusanL82, 2024-05-22:
It worked! No more errors and the system continues to work fine. 
Thank you!

closed 2024-05-22 by SusanL82