open-ephys/plugin-GUI #617: can not see any plugins on the plugin installer

JieZ1230, 2024-07-01:
When I open the plugin installer, I don't see the plugin list. I can't select the plugin I need on the left.

Computers are network connected. How can this problem be solved

[image]

anjaldoshi, 2024-07-01:
Hi - I just verified that both of the services necessary for the Plugin Installer to work are up and running. It could be because your Network Admin has blocked access to certain endpoints/URLs automatically and that's why the Plugin Installer is not able to access its cloud gateway to fetch the data. Can you try to access the following URL directly via your browser using the same machine and see if it loads without any errors for you?
https://open-ephys-plugin-gateway.herokuapp.com

JieZ1230, 2024-07-02:
> Hi - I just verified that both of the services necessary for the Plugin Installer to work are up and running. It could be because your Network Admin has blocked access to certain endpoints/URLs [...]

HI! Thank you for your help. I visited the website you provided, but it could not be opened successfully, and it said "Cannot access this page". How can I solve this problem?

Thank you again！
BEST!

anjaldoshi, 2024-07-02:
That means the site is blocked on your network, and you'll have to whitelist it. If your network is managed by your institution/organization then you'll have to ask your IT team to whitelist the site for you.

JieZ1230, 2024-07-04:
> That means the site is blocked on your network, and you'll have to whitelist it. If your network is managed by your institution/organization then you'll have to ask your IT team to whitelist the site for you.

Thanks very much for your help. The problem has been solved!!

closed 2024-07-11 by anjaldoshi