open-ephys/plugin-GUI #618: Fails to load Acq Board default configuration with the right .so files having been moved to /usr/lib/ (verified to be in LD_LIBRARY_PATH) on Ubuntu

sh00kti, 2024-07-02:
System:
OS: Ubuntu 22.04
Acquisition board: Open Ephys FPGA (not Opal Kelly)

Steps to reproduce:
Start the GUI > File > Load a default config > Select Acq. board > Load > Select Open Ephys FPGA > Immediately error is thrown.

This apparently seems like a known issue on Linux, the Discord community directed me to move .config/open-ephys/shared-api8/libonidriver_ft600.so and .config/open-ephys/shared-api8/libftd3xx.so to /usr/lib/ (which is in LD_LIBRARY_PATH), the issue still persists.

The files to be sent to the support team as directed by the error message pop-up:
[activity_2024-07-02_16-03-40.log](https://github.com/user-attachments/files/16066718/activity_2024-07-02_16-03-40.log)
recoverConfig.xml: https://pastebin.com/ksCNuMrp

Console output:
[image]

anjaldoshi, 2024-07-03:
Can you try the following steps and let me know if it helps?

1. Clear your signal chain (Ctrl + Backspace) and open Plugin Installer (Ctrl + P)
2. Scroll down to OE FPGA Acquisition Board plugin. Click on it and uninstall the plugin. Then close the GUI.
3. Navigate to `~/.config/open-ephys/shared-api8/` and remove the `libftd3xx.so`, `liboni.so`, and `libonidriver_ft600.so` files. Also remove them from `/usr/lib/` if present.
4. Launch the GUI (make sure the signal chain is clear), open Plugin Installer and reinstall the latest version of OE FPGA Acqusition Board plugin. Once installed, close the GUI again.
5. Copy the newly installed `libftd3xx.so` and `libonidriver_ft600.so` files from `~/.config/open-ephys/shared-api8/` to `/usr/lib/`
6. Launch the GUI again, and now drag the OE FPGA Acquisiton Board plugin from the processor list to the signal chain. Wait for the plugin to initialize as it may take a few seconds.

sh00kti, 2024-07-03:
That worked! Thanks!

holmescharles, 2025-06-09:
Hi, I just tried this fix, and while I can now avoid the immediate crash with the suggestion to email the couple of files to you guys, the software cannot find the acquisition board. 

[image]

I have confirmed the OS sees the board:

```
> lsusb

Bus 002 Device 004: ID 0403:601e Future Technology Devices International, Ltd Open Ephys FT600 USB board
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 001 Device 002: ID 413c:2003 Dell Computer Corp. Keyboard SK-8115
Bus 001 Device 005: ID 0bda:b00a Realtek Semiconductor Corp. Realtek Bluetooth 4.2 Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

sh00kti, 2025-06-09:
@anjaldoshi 

[image]

udev rules (couldn't upload file):
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="601e", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="601f", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="602a", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="602b", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="602c", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="602d", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="602f", MODE="0666"