open-ephys/plugin-GUI #619: Open Ephys recording files losing the sample rate information

salinvia-bp, 2024-07-10:
Hi,

We recently have problems opening our recorded .continuous files using Matlab after the version update. According to Matlab, the sample rate is not in the file (whether at the header or the content):
[image]

Here are the settings on the acquisition panel:
[image]
[image]
[image]

Please share your experiences of encountering similar situations and ways to work around this problem.

Many thanks.

medengineer, 2024-07-10:
Hi,

Can you post the contents of the structure.openephys for the recording with the missing sample rate data? 

Which version did you upgrade from and to?

salinvia-bp, 2024-07-11:
Hi ,

Thanks for reaching out.

Here's the contents of the structure.openephys for the specific recording:
 format: 'Open Ephys Data Format'
         version: 0.6000
    header_bytes: 1024
     description: 'each record contains one 64-bit timestamp, one 16-bit sample count (N), 1 uint16 recordingNumber, N 16-bit samples, and one 10-byte record marker (0 1 2 3 4 5 6 7 8 255)'
    date_created: '08-Jul-2024 21:23:17'
         channel: 'CH10'
     channelType: 'Continuous'
     blockLength: 1024
        bitVolts: 0.1950

We are currently using the v0.6.6 software currently. We'll get back to you once we have the older version information examined.

Many thanks,
Lei

medengineer, 2024-07-11:
Thanks for this information. The `sampleRate` missing issue was fixed in `OpenEphysDataFormat v0.6.2-API8` which should be the latest available for install in the GUI. Can you confirm you have `OpenEphysDataFormat v0.6.2-API8` installed via the Plugin Installer (File -> Plugin Installer).
[image]

closed 2025-06-08 by jsiegle